jfedor2/hid-remapper #184: new firmware doesnt work

undecideduser8, 2024-10-23:
flashed the new firmware to the adafruit feather and it doesnt work
had to install the one before because it doesnt detect the mouse

jfedor2, 2024-10-23:
What mouse model are you using?

undecideduser8, 2024-10-23:
> What mouse model are you using?

logitech g502 hero

JostBrand, 2024-10-24:
For me the mouse wheel stopped working. The input is  registered in the monitor tab of the configuration page as:
`Usage 0x00010038 V scroll
Last value 1
Min value -1
Max value 1
`

jfedor2, 2024-10-24:
@undecideduser8 When your mouse wasn't working, did the inputs still show up in the Monitor tab?

hanga61, 2024-10-24:
For me it turns out that "only" usb hub isn´t working. single mouse or keyboard works.

jfedor2, 2024-10-24:
@hanga61 Has it worked for you on the previous release?

hanga61, 2024-10-24:
I´ve tried the june release and it´s also not working.
I have tried three different hubs, but all passive.
Also tried different USB Ports.

Found your project on the weekend and imemdiately ordered two adafruit feather! :) Thanks for your work!

sdixon99, 2024-10-25:
The latest firmware also caused my pico based hid-remapper to stop working.  In my case, the pico is wired into a two port hub directly so that correlates with the report from hanga61.   Going back to the last firmware release restores functionality.  In addition, I put together another pico wired to a single usb (from a cut off usb extension cable) and that one works fine with the latest firmware.  So it does appear that the problem is with the new firmware used with usb hubs.

undecideduser8, 2024-10-25:
> @undecideduser8 When your mouse wasn't working, did the inputs still show up in the Monitor tab?

didnt check that

undecideduser8, 2024-10-25:
> The latest firmware also caused my pico based hid-remapper to stop working. In my case, the pico is wired into a two port hub directly so that correlates with the report from hanga61. Going back to [...]

mine is also plugged to a usb hub

jfedor2, 2024-10-25:
OK, I can replicate the hub issue.

Any other issues with the new release, or existing previously, please file separate issues.

jfedor2, 2024-10-25:
I have a test build that should fix this regression if anyone wants to try ("artifact" link at the bottom):

https://github.com/jfedor2/hid-remapper/actions/runs/11525026520

Please note it doesn't make _all_ hubs work, just the ones that worked before the latest release.

sdixon99, 2024-10-25:
The remapper.uf2 firmware from artifact.zip restores function to my single pico based device with usb hub.  Thanks!

hanga61, 2024-10-27:
Do you have any advice for a working USB hub? Iḿ new to the project and my existing hubs doesn´t work. (Not with the new version, nor an older)

jfedor2, 2024-10-27:
I've had good results with this UGREEN one, but keep in mind that with products like this, silent revisions are not uncommon so there's no guarantee that you will get the same internals when you buy something that looks like this.

[image]
[image]

hanga61, 2024-10-31:
I could borrow one from AUKEY that looks similar to the ugreen and yes, it works.
But from time to time mouse or keyboard stopped working. Sometimes completly Till replug, Sometimes for seconds.
Also then it sends the last State, e. G. Wehen i'm typing it could send lllllllllllllllllllllllllllllllllllllots of letters.

jrhaggerty, 2024-10-31:
@sdixon99 What two port hub do you have wired to the single pico? I actually tried this last week with cheap splitter board/cables and it didn't work.. but I just found this issue was present so may have to try again. Curious what works for you though.