jfedor2/hid-remapper #187: Not an issue, just want to know if this is possible with expressions...

ThePilgrums, 2024-11-11:
Hi, absolutely love the project! A great thing for assistive technology and anyone converting an hid device. I've been looking for a device that can do this.

So anyway, I have duchenne muscular dystrophy and I use a setup involving an XAC and software called xpadder to use it as a mouse. The problem is certain programs/Windows pop ups stop it from functioning, so I've been looking for an arduino/pi pico alternative or adapter.

I read through your expression guide, but my very limited coding knowledge and unfamiliarity with reverse Polish notation didn't get me very far.

Anyway, before I waste any more of your time, can you have an input alternate between two key presses/mouse movements? In my setup I have right bumper set to move the mouse up while held, then the next time you hold the right bumper it moves the mouse down, then it goes back and forth in a loop from there. The right trigger does the same with left and right mouse movement. I'm not sure this matters, but this is all on one "layer" as you call them in your software.

I uploaded a video just in case I didn't explain myself well enough:
[https://youtu.be/OoyuBFAy59Y](https://youtu.be/OoyuBFAy59Y)

Any help would be very much appreciated!

jfedor2, 2024-11-11:
Yeah, something like that should be possible.

You could make a mapping like this:

Expression 1 -> Cursor X

and set Expression 1 to:
```
/* register 1 is either 1 or -1 */
/* meaning the cursor will move right or left */

1 recall
/* register is initially 0 so make it 1 in that case */
dup not add
/* check if button was just pressed */
0x00090007 prev_input_state_binary not
0x00090007 input_state_binary
mul
/* if the button was just pressed, flip the direction */
-1
1
ifte
mul
/* store the value for next time */
1 store

/* apply the direction to the movement */
0x00090007 input_state_binary
1 recall
mul
```

And a similar thing for Cursor Y. Here's a JSON file with the complete config: [hid-remapper-flip-mouse.json](https://github.com/user-attachments/files/17707185/hid-remapper-flip-mouse.json)

You would of course have to update it with the specific buttons you want to use and possibly adjust sensitivity etc.

I know the expression syntax isn't the most user friendly thing. Let me know if you need any clarification or have some other ideas.

jfedor2, 2024-11-11:
Actually this should be possible even without using expressions at all, but I'm not sure if such a configuration is necessarily easier to understand.

It uses layers 1-4 and the web configuration interface currently only shows layers up to 3 (they go up to 7 internally).

But if anyone's curious here's the JSON file: [hid-remapper-flip-mouse2.json](https://github.com/user-attachments/files/17707577/hid-remapper-flip-mouse2.json)

And here's what it looks like with Layer 4 photoshopped into where it should be:
[image]

ThePilgrums, 2024-11-13:
Oh, thank you!! I'll try it as soon as I can!

closed 2024-11-13 by ThePilgrums