jfedor2/hid-remapper #191: Delay periods

RuloH, 2024-11-29:
Is it possible to set delay periods in between key strokes? For example F1 and wait one second  then press left mouse button.

jfedor2, 2024-11-29:
Not explicitly, but you could set the macro entry duration to, say, 100ms, and then define the macro as:

F1-Nothing-Nothing-Nothing-Nothing-Nothing-Nothing-Nothing-Nothing-Nothing-Left button

And of course anything is possible with expressions.

RuloH, 2024-12-01:
My proficiency in coding is very limited and I haven't been able to figure out the expression method. Could you please point me in the right direction where can I study the subject? Thank you very much for your awesome work.

RuloH, 2024-12-01:
I just don't get the expression.. In this example:

0x00080002 input_state_binary time 600 mod 200 gt mul

if i use it with a momentary button. How do I make it toggle on or off when pressing , so I don't have to hold down the button to keep the loop running?

jfedor2, 2024-12-06:
There are a few ways you can do this, you can have an expression that just says:
```time 600 mod 200 gt```
This will just "blink" regardless of button presses.
Then you can create a mapping like this:
Expression 1 -> Some button
and make it only active on layer 1.
Then create a mapping that will activate layer 1 when some key is pressed and deactivate it when the key is pressed again:
Some key -> Layer 1, with the sticky flag enabled.

Or you could implement everything yourself, let's say that register 1 will keep the "turbo enabled" state.
Then you can flip it every time some button is pressed, let's say Button 1:
```
0x00090001 input_state_binary
0x00090001 prev_input_state_binary not
mul
1 recall not
1 recall
ifte
1 store
```

The first three lines detect when the button was just pressed (previous state=0, current state=1). Then we use that value to choose between the inverted value of register 1 or just the current value of register 1 (no change). Then we store it back into register 1. This is just a manual implementation of the sticky flag behavior on mappings. Then you can use this value to make something "blink":
```
time 600 mod 200 gt
1 recall
mul
```

RuloH, 2024-12-07:
Genius! thank you very much!

Machforr, 2025-01-25:
I am having issues with delays as well, in that I can't set them.
Over a remote connection, the macros are too fast to be registered by the remote computer and just don't work as they should.
For example, 'ctrl+c' just gets registered as 'c'... or sometimes just the 'ctrl' key (based on a keylogger).
I tried to look at expressions to add delays, but I can't figure out how they work, not being a programmer.

For example, press the CTRL key, then 100ms later press the C key for +/- 075ms, then release the C key then the CTRL key.
That would solve all my problems with lag over remote access... just to give the system time to register the key being pressed.

Any help would be greatly appreciated :)

jfedor2, 2025-01-25:
@Machforr What do your macros look like?

If you set them like this, I think it should work:

[image]

Or even like this if you want, though I don't think it will make a difference:

[image]

Machforr, 2025-01-25:
*Slaps forehead*
Of course it had to be the combination I didn't try. I was giving a shot at RPN expressions with chatGTP and it was getting obscure fast  :)

Thanks so much!