jfedor2/hid-remapper #192: String processing - order of characters in output string is changed

konik-sk, 2024-11-29:
Hi, I have been testing HID remapper to remap output from a card reader. In my case the card reader generates string consisting of only numbers separated by slashes. Whole length of string is my case 16 characters including delimiters (e.g., 1234/5678/9/1234).

Problem is that order of chars in the output (default setting, no mappings) is changed. Always changed in the same way. I have tried single and dual pi version, but always with the same output. (e.g., /24562/8719/1433). I have tried different polling settings, but without success.

Any idea what may be wrong? Thanks

jfedor2, 2024-11-29:
I'm not sure what's going on, any chance you can plug the reader into a PC, do a Wireshark dump and post that? Ideally with the report descriptor. Maybe there's something weird there that gets interpreted differently by HID Remapper.

konik-sk, 2024-11-30:
After checking dump was realized that card reader is sending string in three "batches", where all "keys" are pressed at once. After explanation from jfedor2 I realized that HID Remapper is passing those keys in different representation (respecting those keys pressed) but intentionally not distinguishing order among them, as they are all pressed at once.  -> issue closed.

closed 2024-11-30 by konik-sk