jfedor2/hid-remapper #193: HID remapper doesn't monitor any game controller except mouse  and keyboard

pftmclub, 2024-11-30:
Hello, i'm trying to remap my GP2040 controller, but none of them working . Tried it with keyboard and it's work successfully, does  anyone know this  problem?

Gamepad:

https://github.com/user-attachments/assets/c69d0429-65f2-4e99-80c4-f71d3a473d2d

Keyboard:

https://github.com/user-attachments/assets/e6717dcd-671b-4fcc-85de-6fc36c832ff7

jfedor2, 2024-11-30:
I just tested that it works in Switch mode for example.

HID Remapper doesn't care about the shape of the plug as long as the data lines are wired properly. Does your keyboard work with the same cable?

pftmclub, 2024-11-30:
> I just tested that it works in Switch mode for example.
> 
> HID Remapper doesn't care about the shape of the plug as long as the data lines are wired properly. Does your keyboard work with the same cable?
My  keyboard have usb cable built-in ,

I tried to plug the USB-C directly into the usb hud and it's flashing. The other case is plug the Printer cable (USB-B) , nothing seem work .

USB-C:
[image]

USB-B :
[image]

jfedor2, 2024-11-30:
So it works when you connect it directly, but doesn't work when it goes through the Neutrik-or-whatever-style port on the enclosure? I don't see why it wouldn't.

pftmclub, 2024-12-01:
> So it works when you connect it directly, but doesn't work when it goes through the Neutrik-or-whatever-style port on the enclosure? I don't see why it wouldn't.

Yeah, and its strange that the controller still function over OLED screen but the remapper doesn't recognize it.
To put it simple , i plug the cable to usb-c to usb-b hud , then using the usb-b cable into usb hud of the pico board