jfedor2/hid-remapper #195: Ability to add two joysticks?

MGlaab, 2024-12-14:
I've followed your tutorial on YouTube explaining how to use one joystick with HID Remapper. I have some noise readings but I'm reading your examples with expressions to mitigate some of the jitters.
I also saw your video using two joysticks with two Remappers plugged into a USB hub.
My question is would it be worth adding I2C support so you could use the Qwicc (https://www.sparkfun.com/qwiic) ecosystem of sensors and breakouts? I'm specifically thinking of an ADC.

MGlaab, 2024-12-14:
It looks like I could get two joysticks working using the Adafruit Feather USB Host board. The RP2040 chip has 4 analog pins available on that board.

jfedor2, 2024-12-14:
You'd have to compile a version of the firmware with NADCS set to 4, but yeah, it should work.

MGlaab, 2024-12-14:
Would I add this line:
#define NADCS 4
to this file:
firmware/src/boards/feather_host.h
or is NADCS defined elsewhere for the Adafruit board?

I used your remapper_v8.h for reference.

jfedor2, 2024-12-14:
Yeah.

Also define ADC_ENABLED and probably remove bits 26-29 from GPIO_VALID_PINS_BASE.

MGlaab, 2024-12-15:
Ok, I added:
#define ADC_ENABLED
#define NADCS 4

and I see:
GPIO_VALID_PINS_BASE 0b00111111000000001101111111111111

but to be honest you lost me about removing bits. I understand that GPIO_VALID_PINS_BASE sets up the states of the pins, but where would I find out which one or zero corresponds to the pins 26-29?

MGlaab, 2024-12-15:
old line:
#define GPIO_VALID_PINS_BASE 0b00111111000000001101111111111111

new line:
#define GPIO_VALID_PINS_BASE 0b00100001000000001101111111111111

Basically, I counted from the right side to the left, starting from 1, to 26-29, and changed the value in those indexes from 1's to 0's. Is this correct or is the index shifted by 1 because you start from index 0?

jfedor2, 2024-12-15:
Yeah, it starts at 0 so it should be:
```
#define GPIO_VALID_PINS_BASE 0b00000011000000001101111111111111
```
That's just so it doesn't try to use the analog pins as digital for buttons.

MGlaab, 2024-12-15:
Ok, thank you. I'll test this out and post an update when I get it working.

MGlaab, 2024-12-19:
I'm having an issue compiling a version of the firmware with the Feather's ADC's enabled. I'm using your suggestion, I forked the repo and am using Github actions. the main.cc file is looking for a hardware/adc.h file. I'm unable to find the hardware folder or the adc.h file. Please advise.

jfedor2, 2024-12-19:
Ah, it seems I have hardcoded including the hardware_adc library in CMakeLists.txt to the remapper_v8 board name:
```
$<$<STREQUAL:${PICO_BOARD},remapper_v8>:hardware_adc>
```
Add a similar line there for the Feather or just include it unconditionally like the rest of the libraries there.

I will figure out a more elegant way to do it eventually.

MGlaab, 2024-12-19:
Thanks. I added a line for the Feather.
```
$<$<STREQUAL:${PICO_BOARD},feather_host>:hardware_adc>
```
I built the firmware successfully.