jfedor2/hid-remapper #196: Compatibility with Waveshare devices

scatterthought, 2024-12-18:
I see from other discussions that the Waveshare RP2040-Zero works with the remapper.uf2 firmware. Is it likely that it will also work with the RP2040-Zero that has its USB-C port on a separate adapter board? I'm guessing that it should, since the hardware appears otherwise identical.

https://www.waveshare.com/product/raspberry-pi/boards-kits/raspberry-pi-pico-cat/rp2040-tiny.htm?sku=24665

I'm also wondering if you'd expect it to work with the RP2040-One board, which has a USB-A port instead of USB-C and 4GB of Flash memory.

https://www.waveshare.com/product/raspberry-pi/boards-kits/raspberry-pi-pico-cat/rp2040-one.htm

Apologies if these are silly questions. I'm new to RP2040s and I'm not a software developer, so I don't know what to expect. If you haven't tested these boards and think there's value, I'm happy to pick up a couple to try out. Perhaps we could then add a table of compatible devices to the Pico webpage.

Thanks for your fantastic work!

jfedor2, 2024-12-18:
I haven't tested personally, but they should both work, the type of plug that goes to the PC shouldn't matter here.

scatterthought, 2024-12-18:
Cool, I'll give them a shot. I'm thinking of fitting a board into the chassis of my Trackman Marble and then having a detachable USB-C cable.

MGlaab, 2024-12-19:
I can confirm that remapper.uf2 firmware works on both of the Waveshare boards(RP2040-Zero, RP2040-One).

scatterthought, 2024-12-19:
Thanks for chiming in! I've been going through my options and now I'm leaning back toward the basic RP2040-Zero. I can get a three-pack for CAD$20 and it seems like the most versatile one to have handy.

scatterthought, 2025-01-02:
Happy new year!

I've got HID Remapper working beautifully with an RP2040-Zero, but I've noticed that the LED doesn't blink when the board registers input (since the Zero uses a neopixel). Would it be easy/possible to add this feature so that the behaviour matches the Pico?

I found a couple of posts about using the neopixel, but I don't have the programming expertise necessary to adapt it myself.

https://www.arduinoetal.net/?q=node/760
https://www.reddit.com/r/raspberrypipico/comments/x3dfjx/the_ws2812_din_rgb_on_waveshare_rp2040_zero/

No big deal if you'd rather not, but I figured it can't hurt to ask. It threw me off a little that the board wasn't blinking, so I wasn't sure if it was working properly at first.

Cheers!

andreapede, 2025-01-02:
I confirm that the led is always on and super bright.
Could be nice having some blinking for it.

scatterthought, 2025-01-02:
Well, that's odd. For me the LED is always off, unless I press the RESET button.