jfedor2/hid-remapper #197: Abilty to change hid-renappers output descriptor, vid, pid

djsedaw, 2025-01-04:
I have an HP omen 1100 keyboard, no problem connecting and using hid remapper with it but it shows up in windows as a hid remapper keyboard which means the HP software to change the setting on the keyboard no longer work as i think it no longer sees it as a omen 1100,  is there a way the change the output descriptor, vid, pid to the omen 1100 descriptor, vid, pid?

jfedor2, 2025-01-04:
You can change the VID, PID and descriptor, but it won't work the way you think, the software from HP won't be able to configure your keyboard when it's plugged into HID Remapper.

djsedaw, 2025-01-04:
Ok thanks, will have to plug it straight in if i want to change  a setting

taneofpain, 2025-02-09:
i see in custom usage how to change VID and PID. How do you change the descriptor (ie. the text name).

jfedor2, 2025-02-09:
To change VID/PID or the device name you have to compile a custom build of the firmware.

taneofpain, 2025-02-10:
ok thank you. mind sharing some advice on which file has that information in it to edit?  For context i'm trying to use a gamo2 be2ps device with a yuancon controller in infinitas mode but doesn't work as the controller isn't supported.  my theory is the be2ps device looks not just at the VID/PID (which are the same in this device as for other models it's supports) but also at the descriptor which is different so i'm try to spoof one it knows about to see what happens.

jfedor2, 2025-02-10:
By default unmapped inputs passthrough is enabled, but it doesn't mean it passes everything through transparently. It still behaves only as the device selected as the emulated device type. If whatever you plug in happens to have the same inputs then they will be passed through, but if your adapter expects a certain format of the data because it only works with specific devices then it likely won't work.