jfedor2/hid-remapper #199: Web config - Your browser doesn't support WebHID.

kibernaut, 2025-01-08:
Hi!

I'm trying to set up a web config on a local web server (Laragon, Windows). But the configurator says "Your browser doesn't support WebHID." I tried it in different Chrome-based browsers (Brave, Chrome), but the result is the same in all of them.

[image]

The original configurator (https://www.jfedor.org/hid-remapper-config/) works without any problems in these browsers.

Is it even possible to run it in such a configuration?

jfedor2, 2025-01-08:
Since WebHID is a sensitive feature from a security point of view, Chrome only enables it on pages served over HTTPS.

Though I've noticed that it also works without HTTPS when the server is localhost.

kibernaut, 2025-01-08:
Thank you very much.

I had my suspicions about HTTPS, but in the absence of experience I somehow did not think of checking via localhost. I confirm (see screen), when accessing via localhost it worked without any problems.

[image]