jfedor2/hid-remapper #203: PC acting as the HID keyboard for adaptive gaming

ehrhartd, 2025-01-17:
I'd like to have the PC act as the HID keyboard to emulate an Xbox controller.  This would wind up connecting to an Xbox Adaptive Controller via USB. Could this just be done on the PC side to get the keyboard strokes to feather board or would the hid remapper code need to be modified to recieve signal from the PC?

jfedor2, 2025-01-18:
There's no way to do it currently, but it's something I'm thinking about.

Keep in mind there is no direct Xbox compatibility so to connect it to an Xbox console you'd have to use the XAC or some adapter.

jfedor2, 2025-01-24:
Maybe this will be of interest to you:

https://github.com/jfedor2/hid-forwarder

It lets you generate inputs on the PC using any method and send them to a receiver that can emulate all the device types HID Remapper can.

controllercustom, 2025-02-13:
@ehrhartd @jfedor2

The usbnswa project may be related to what you are considering. An ESP32-S3
emulates a Nintendo Switch gamepad on its USB OTG port. The ESP32-S3 serves
a web page which look like a virtual gamepad. Clicks on the web page buttons
are relayed to the ESP32-S3. The ESP32-S3 converts the clicks into USB HID
gamepad button presses going out to the Switch.

Changing the USB HID gamepad can make it compatible with the XAC instead of
the Switch. hid-remapper support several different gamepads as well as mouse
and keyboard.

Chrome - Computer -WiFi- ESP32-S3 - USB OTG - Nintendo Switch or XAC

hid-remapper does not run on ESP32-S3 but maybe it can run on pi picow.
A pi picow can do the same web server stuff, I think, as the ESP32-S3.

One disadvantage for pi picow is sellers limit the number of boards per order.

If the computer has eye gaze hardware, the user can use eye gaze to click
on the web page buttons. This is an assumption since I have never used
eye gaze hardware but have seen videos.

https://github.com/touchgadget/usbnswa