jfedor2/hid-remapper #204: Using with nrf52840 Dongle + default bootloader

eyJhb, 2025-01-21:
What is the possibility of using this with a default NRF52840 Dongle, which has the default bootloader? As far as I can read, if I want to get them to work, I need to flash the bootloader before I can flash the uf2 files.

Any possibility of generating the basic hex file, that can be flashed using the default bootloader?

eyJhb, 2025-01-21:
I just tried building it with `nrf52840dongle_nrf52840`, but once I flashed it the board shows no signs of life. E.g. no LEDs, nothing in dmesg, etc. 

It might need some additional configuration changes.

eyJhb, 2025-01-21:
> Sorry, what fork?

Yeah, sorry, it would _greatly_ help you if I actually had pasted the link like I thought I had :) This one https://github.com/DeltaWhy/hid-remapper/tree/dongle-nrf52840 , I've made a couple of changes to make the LEDs, buttons, etc. work. but it very much feels like it's just a dirty fix that doesn't really aid anything. 

I have a sneaking suspicion, that the reason it isn't working, might be alike this issue here https://devzone.nordicsemi.com/f/nordic-q-a/117767/nrf_desktop-nrf52840dongle-no-signs-of-life-after-flash .

Also, sorry for the low quality replies, I'll try to do better. I really appreciate your prompt replies :)

jfedor2, 2025-01-21:
What dongle do you have? The fork seems to be for the Makerdiary one (which ships with a UF2 bootloader as far as I can tell), while I was thinking of the Nordic one the whole time.

What device are you trying to connect to?

eyJhb, 2025-01-21:
I have the nRF52840 Dongle ( https://www.nordicsemi.com/Products/Development-hardware/nRF52840-Dongle ), but for some reason using that `.hex` file (the one I build using the fork, using the target `dongle_nrf52840`) got me more (device showing up as hid, etc), than using `nrf52840dongle_nrf52840` as the target board.

Keep in mind, I'm always flashing the hex files located in `./firmware-bluetooth/build/zephyr/remapper.hex` when testing this.

Sadly, I don't have any way to debug the nrf52840 dongle, my only debugging alternative is using the nrf52840 dk, which wouldn't really help.

jfedor2, 2025-01-21:
Maybe try disabling the CONFIG_USE_DT_CODE_PARTITION option in prj.conf? I don't really have other ideas what could be stopping it from working using the `nrf52840dongle_nrf52840` board definition.

What device are you trying to connect to it? Are you sure it's Bluetooth LE?

eyJhb, 2025-01-22:
I've done some further testing, and I ended up getting it to work after much trial and error.
Luckily, the changes needed to get it to work are _minimal_, at least at a basic level.

I had to change the `CONFIG_USE_DT_CODE_PARTITION=n`, as well as `CONFIG_NVS=n` (uncommented in theory).
The changes are listed here below.

```patch
diff --git a/firmware-bluetooth/prj.conf b/firmware-bluetooth/prj.conf
index 99e3220..22144fa 100644
--- a/firmware-bluetooth/prj.conf
+++ b/firmware-bluetooth/prj.conf
@@ -3,7 +3,7 @@ CONFIG_LOG_MAX_LEVEL=3
 CONFIG_LOG_FUNC_NAME_PREFIX_INF=y
 CONFIG_LOG_BUFFER_SIZE=4096
 
-CONFIG_USE_DT_CODE_PARTITION=y
+CONFIG_USE_DT_CODE_PARTITION=n
 CONFIG_KERNEL_BIN_NAME="remapper"
 CONFIG_BUILD_OUTPUT_UF2=y
 
@@ -18,7 +18,7 @@ CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=16384
 CONFIG_FLASH=y
 CONFIG_FLASH_PAGE_LAYOUT=y
 CONFIG_FLASH_MAP=y
-CONFIG_NVS=y
+# CONFIG_NVS=y
 CONFIG_SETTINGS=y
 
 CONFIG_GPIO=y
```

However, with that configuration, the LEDs will not light up with the correct colors.
The "blue" LED will be red (led1), and the "red" LED will be green (led0), but this should be easily fixable.

Is there any possibility of adding the nRF52840 Dongle as a prebuilt image as well? Maybe others will benefit from it.

As a side-note, I built it using the following steps

```
docker run --rm -v $(pwd):/workdir/project -w /workdir/project/firmware-bluetooth nordicplayground/nrfconnect-sdk:v2.2-branch west build -b nrf52840dongle_nrf52840

# the below can be ran inside the docker container, but it requires that the following plugin is installed for nrfutil
# nrfutil install nrf5sdk-tools

nrfutil pkg generate --hw-version 52 --sd-req=0x00  --application firmware-bluetooth/build/zephyr/remapper.hex --application-version 1 remapper.zip 
nrfutil dfu usb-serial -pkg remapper.zip -p /dev/ttyACM0
```

Once again, thanks for this awesome project, and your aid in resolving this.
I only wish that I knew about this project, a couple of weeks before, as I just adapted the `nrf_desktop` application ([here](https://github.com/eyJhb/sdk-nrf/commits/glove80-hid-forward-v2.9.0/)), to function as a HID forwarder.
However, the killer feature I missed, is actually reading the report map, and using that.

scatterthought, 2025-01-29:
I'd agree that a prebuilt image for the MDK USB dongle would be great. It's easier to find those in Canada than the Adafruit or Seed Studio boards, and the optional case would be great for people who just want a prebuilt solution.