jfedor2/hid-remapper #205: waveshare rp2040 zero not working with remapper_waveshare_rp2040_pizero.uf2  Firmware

andreapede, 2025-01-24:
I'm using a waveshare RP2040 zero board connected with the usb directly to an USBC external cable.
If I use the basic remapper firmware everything works perfectly out of the box
The only cosns is the led that it is always on, but also it is not a normal led but a neopixel ws2812

While if I use the remapper_waveshare_rp2040_pizero.uf2, there is no way to work it properly: i can connect to the web customizer and stop. 
The only message i have in the monitor when i plug the keyboard is a change in GPIO1 but no keypress or else.
Am I missing something?

jfedor2, 2025-01-24:
> While if I use the remapper_waveshare_rp2040_pizero.uf2, there is no way to work it properly: i can connect to the web customizer and stop.

This file is for a different board, the [RP2040-PiZero](https://www.waveshare.com/wiki/RP2040-PiZero). Yes, their naming scheme is confusing. There is no special build for the RP2040-Zero.

andreapede, 2025-01-25:
Oh. I'm sorry for the confusion. thanks for the reply