jfedor2/hid-remapper #207: Utilizing custom usages for device input buttons that don't appear in the monitor

nupedogg, 2025-02-02:
Hi, I am trying to use the hid-remapper with the Contour ShuttlePRO v2 input device, which does not have onboard memory, to be able to use the device portably on multiple workstations. Nearly all the physical buttons appear on the Monitor when clicked, with the exception of two buttons. I can extrapolate what their Usage ID would be based on the other buttons that do show up in the Monitor (0x00090001 through 0x009000d), and want to see if I can force them to map. 

After I have created the custom usages in Settings, how do I map those custom usages? I don't see them as available input options under mappings, nor is there a "+" button to click on a similar fashion as the Monitor to begin creating a new mapping with the custom usage. Below is the custom usage I created:

[image]

Thank you so much for your help, I can already see how useful your device will be in corporate settings where software installations are not allowed on user workstations.

jfedor2, 2025-02-02:
Use these custom usages for the Shuttle Pro v2 for the two black buttons to work (also included are usages for the wheel and jog dial, which are incorrectly marked as relative in the device's descriptor). If you define them correctly they will show up in the Monitor. You can just import the attached JSON file.

[image]

[hid-remapper-shuttle-custom-usages.json](https://github.com/user-attachments/files/18634566/hid-remapper-shuttle-custom-usages.json)

nupedogg, 2025-02-02:
Thank you so much! Everything maps properly now. Greatly appreciate your efforts in creating this and responsiveness.

jfedor2, 2025-02-06:
You can map 0xffff0038 to V scroll and it should work. You might want to set the scaling to 0.001. I got reports from another user that it still goes to fast with this setting so we might have to come up with some workaround as the internal precision is 0.001. But give that a go.

nupedogg, 2025-02-06:
I tried that, with a few observations:
- the speed is quite fast, and does not correspond to the amount the dial is turned
- after the jog dial springs back to the center, the scrolling also springs backward. I'm using this for radiology, scrolling through a stack of images, and if I use this to scroll down to the bottom, it springs back to the top with as the jog dial springs back 
- I think the spring back to the center also seems to introduce additional key strokes or commands that gum up the system (I hear some beeps that indicate that additional commands are being sent, and if I use a mouse to move the cursor, it moves extremely slowly and is essentially unresponsive until I unplug the HID remapper from the USB port.

jfedor2, 2025-02-08:
Maybe try this. Make an expression like so:
```
time 100 mod 0 eq
0xffff0038 input_state
mul
```

And map that expression to "V scroll".

It sort of applies a 0.01x pre-scaling by only taking the input every hundredth frame. So the scaling parameter should have some usable values now in the valid range.

nupedogg, 2025-02-09:
That does allow for a usable scroll, but interestingly with a fixed speed of scrolling through a stack of radiology images regardless of how much the jog dial (ffff0038) is turned. In Contour's software, it looks like there are 7 levels of control for each direction the dial is turned:

[image]

I have configured that to change the frequency of mouse wheel down or up to 2, 5, 10, 15, 30, 45 times per second, and as fast as possible for "Shuttle in 1, 2, 3, 4, 5, 6, and 7). A couple of representative images below. Is it possible to simulate that type of behavior

[image]

[image]

For the Jog Wheel (ffff0037), is it possible to program different actions turning it left versus right? I use the Jog Left and Jog Right to simulate the left arrow and right arrows, respectively.

[image]

jfedor2, 2025-02-09:
For the inner wheel you can use an expression like this:
```
0xffff0037 input_state
0xffff0037 prev_input_state sub
dup
128 gt
-256 mul
swap dup
-128 lt
256 mul
add add
```
And then map that expression to one key with scaling set to 1 and to another key with scaling set to -1.

I'm not sure why it doesn't give you variable speed of scrolling, I don't have the actual device in hand, but I will try and figure this out. Does it also not give you variable speed in other apps, like a browser?

nupedogg, 2025-02-09:
That worked perfectly for the inner jog wheel.

My apologies regarding the jog dial, I was trying it in a browser-based image viewer and was not getting variable scrolling speed, but I just tried it on my real image viewer and it does provide variable scrolling speed. In the monitor, the values range from -7 to +7, correlating to the degree the dial is turned in either direction.

[image]

I think this is now working as intended. Thanks so much for your efforts on this, greatly appreciated!

Zedibee, 2025-04-20:
Hello Nupedogg, another shuttle owner here trying to figure out how to map the outer jog dial/wheel. I've gotten the JSON file and put in both expressions into the expressions fields but still haven't had much luck with getting the outer jog wheel to work as intended on PACS (Fuji). Would you be willing to share any more details regarding your settings on your mappings page? I would just really appreciate any extra help you can lend =D Thanks and best wishes.

nupedogg, 2025-04-21:
[contour-hid-remapper-config.json](https://github.com/user-attachments/files/19834432/contour-hid-remapper-config.json)

Attached is the JSON file I am using that seems to work. I will say that the experience using the wheel to scroll just doesn't perfectly replicate a mouse, though it does provide a respite from scrolling when needed.

slamminsammysj, 2025-10-31:
I also cannot figure out how to map buttons 14 and 15. Can you explain how you were able to extrapolate their position?