jfedor2/hid-remapper #209: Logitech/Saitek X52 Flight joystick hat switch values

controllercustom, 2025-02-12:
The X52 look great in monitor except for the hat switch.

On other joysticks the hat switch value range from 0 to 7 but the X52 uses 1 to 8. 0 represents no direction. However, hid-remapper interprets 0 as up/north.

```
0x09, 0x39,        //     Usage (Hat switch)
0x15, 0x01,        //     Logical Minimum (1)
0x25, 0x08,        //     Logical Maximum (8)

|0x00010039 |               |0  |0  |8  
|0xfff90004 |D-pad down     |0  |0  |1
|0xfff90002 |D-pad right    |0  |0  |1
|0xfff90001 |D-pad left     |0  |0  |1
```

```
Bus 001 Device 013: ID 06a3:075c Saitek PLC X52 Flight Controller
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x06a3 Saitek PLC
  idProduct          0x075c X52 Flight Controller
  bcdDevice           21.00
  iManufacturer           1 Logitech
  iProduct                2 X52 H.O.T.A.S.
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     119
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval              10
Device Status:     0x0000
  (Bus Powered)

 05 01 09 04 A1 01 09 01 A1 00 09 30 09 31 15 00
 26 FF 07 75 0B 95 02 81 02 09 35 15 00 26 FF 03
 75 0A 95 01 81 02 09 32 09 33 09 34 09 36 15 00
 26 FF 00 75 08 95 04 81 02 05 09 19 01 29 22 15
 00 25 01 95 22 75 01 81 02 75 02 95 01 81 01 05
 01 09 39 15 01 25 08 35 00 46 3B 01 66 14 00 75
 04 95 01 81 42 05 05 09 24 09 26 15 00 25 0F 75
 04 95 02 81 02 C0 C0

0x05, 0x01,        // Usage Page (Generic Desktop Ctrls)
0x09, 0x04,        // Usage (Joystick)
0xA1, 0x01,        // Collection (Application)
0x09, 0x01,        //   Usage (Pointer)
0xA1, 0x00,        //   Collection (Physical)
0x09, 0x30,        //     Usage (X)
0x09, 0x31,        //     Usage (Y)
0x15, 0x00,        //     Logical Minimum (0)
0x26, 0xFF, 0x07,  //     Logical Maximum (2047)
0x75, 0x0B,        //     Report Size (11)
0x95, 0x02,        //     Report Count (2)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x09, 0x35,        //     Usage (Rz)
0x15, 0x00,        //     Logical Minimum (0)
0x26, 0xFF, 0x03,  //     Logical Maximum (1023)
0x75, 0x0A,        //     Report Size (10)
0x95, 0x01,        //     Report Count (1)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x09, 0x32,        //     Usage (Z)
0x09, 0x33,        //     Usage (Rx)
0x09, 0x34,        //     Usage (Ry)
0x09, 0x36,        //     Usage (Slider)
0x15, 0x00,        //     Logical Minimum (0)
0x26, 0xFF, 0x00,  //     Logical Maximum (255)
0x75, 0x08,        //     Report Size (8)
0x95, 0x04,        //     Report Count (4)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x05, 0x09,        //     Usage Page (Button)
0x19, 0x01,        //     Usage Minimum (0x01)
0x29, 0x22,        //     Usage Maximum (0x22)
0x15, 0x00,        //     Logical Minimum (0)
0x25, 0x01,        //     Logical Maximum (1)
0x95, 0x22,        //     Report Count (34)
0x75, 0x01,        //     Report Size (1)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x75, 0x02,        //     Report Size (2)
0x95, 0x01,        //     Report Count (1)
0x81, 0x01,        //     Input (Const,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x05, 0x01,        //     Usage Page (Generic Desktop Ctrls)
0x09, 0x39,        //     Usage (Hat switch)
0x15, 0x01,        //     Logical Minimum (1)
0x25, 0x08,        //     Logical Maximum (8)
0x35, 0x00,        //     Physical Minimum (0)
0x46, 0x3B, 0x01,  //     Physical Maximum (315)
0x66, 0x14, 0x00,  //     Unit (System: English Rotation, Length: Centimeter)
0x75, 0x04,        //     Report Size (4)
0x95, 0x01,        //     Report Count (1)
0x81, 0x42,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,Null State)
0x05, 0x05,        //     Usage Page (Game Ctrls)
0x09, 0x24,        //     Usage (Move Right/Left)
0x09, 0x26,        //     Usage (Move Up/Down)
0x15, 0x00,        //     Logical Minimum (0)
0x25, 0x0F,        //     Logical Maximum (15)
0x75, 0x04,        //     Report Size (4)
0x95, 0x02,        //     Report Count (2)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              //   End Collection
0xC0,              // End Collection

// 119 bytes
```

jfedor2, 2025-02-12:
Technically that's on me because it declares the logical minimum/maximum correctly and I just ignore it, but if you want to make it work, check out the "Xbox controller (Bluetooth) adapter for Switch" config in the Examples tab, as Xbox controllers in Bluetooth mode also use 1-8 range for the dpad.

jfedor2, 2025-02-12:
And thanks for the reports, this is all useful information!

controllercustom, 2025-02-13:
Thanks, I will try the Xbox example!