jfedor2/hid-remapper #211: Trouble using XAC with hid-remapper to map a mouse or keyboard to game controller on Xbox

MackNNations, 2025-02-17:
I'm trying to get a usb keyboard or usb mouse to be remapped to game controller output on an Xbox.
I'd like to use mouse and/or keyboard in a game that does have those as normal inputs.
I was hoping hid-remapper would allow remapping to output game controller movements to xbox.
Is this not possible?

I have an XAC with the firmware updated.   I updated to latest feather rp2040 firmware on hid-remapper.  
I set input map to Left Arrow or Cursor X and output to Left Stick.    Nothing happens when I plug hid-remapper into left usb on XAC with a mouse or keyboard attached.    I see red blinking on feather board when I press a key or move mouse, but I get no movement in a game.

Also, when I use the XAC left usb port, the player drifts, with no control.  When I use XAC right usb, I get nothing. 

Am I missing something, or am I just not understanding what hid-remapper is and how it works?

jfedor2, 2025-02-17:
For mouse-to-stick mapping there's an example configuration in the Examples tab that you can use as a starting point. It's called "mouse to analog stick". Just remember to change the emulated device type (in the Settings tab) to "XAC/Flex compatible".

For key-to-stick mappings you can use a configuration like this:
[image]
The scaling parameter determines the position of the stick that the key will map to. "0" is left/top, "255" is right/bottom.

To get rid of the drift, disable the "Unmapped inputs passthrough" setting. It is already disabled in the "mouse to analog stick" example configuration mentioned above so if you load that you should be good.

Plug HID Remapper into the port on the left side of the XAC. If you use the right port it will work, but left/right stick will be swapped and some buttons won't match the labels.

MackNNations, 2025-02-18:
I'm getting some response from WASD keys.  But still have a hard left drift.    I reset it twice using the example, added the scaling, emulation is XAC, and made sure the passthrough boxes were unchecked.  

[image]
[image]

MackNNations, 2025-02-18:
[image]

MackNNations, 2025-02-18:
My mistake.   I think the scaling defaulted to 1 for W and A. 

 I reset them to 0 and the WASD is working properly with no drift.

Can I use a usb hub and plug it into the huid-remapper?   I'd like to plug in the mouse and keyboard into it at the same time, or do I need two hid-remappers?

Thank you for your help.

MackNNations, 2025-02-18:
The hub I tried so far didn't work with mouse and keyboard plugged into hub, then the hub plugged into hid remapper. 
I'll try more hubs if I have any.   

Is there a multi-port (like a built in hub) hid remapper?  I thought you showed one in one of your videos.
Or, would I use a hub plugged into XAC, then use a separate hid remapper for each device in the hub?