jfedor2/hid-remapper #212: Griffin Powermate (The really old version) acts very weird despite the axis being detected.

relink2013, 2025-02-18:
I have one of the old USB Griffin Powermate knobs, I got it probably sometime around 2005 or so. The drivers were long since discontinued, and by default when connected to my Mac the device does absolutely nothing. On MacOS I am able to use a program called [USB Overdrive](https://www.usboverdrive.com/downloads/) to get it working and I can program it to do pretty much whatever I want. I've been using it to scrub the timeline when editing video, it's fantastic!

Im trying HID Remapper on a RP2040-Zero and when I go to try and get the functions through the monitor page this is what I get, which makes sense, at least in my uneducated view:
```
0x00010033         L2 axis         1   -7    7
0x00090001      Left button        0    0    1
```
The problem is that I cant map the `L2 axis` properly. Just as a test I mapped it to `V Scroll` and when I turn the knob to the right it shoots all the way to the bottom of the page very quickly and gets stuck there. I cant scroll back up, not even with my regular mouse. I have to disconnect the knob to use my regular scroll wheel. And scrolling up doesn't work at all. 

Here is some relevant device info I got from USB Overdrive, I'm sure it's probably helpful in some way:
```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>connection</key>
	<integer>31503</integer>
	<key>devicePage</key>
	<integer>12</integer>
	<key>deviceType</key>
	<integer>4</integer>
	<key>deviceUsage</key>
	<integer>1</integer>
	<key>hidDesc</key>
	<data>
	BQwJAaEBBQkZASkBFQAlAXUBlQGBAnUHlQGBAQUBCTMVgSV/dQiVAYECBQwJARUAJv8A
	dQiVBIECCQF1CJUBkQIJAXUIlQixAsA=
	</data>
	<key>interfaceNum</key>
	<integer>0</integer>
	<key>notification</key>
	<integer>23571</integer>
	<key>numControls</key>
	<integer>6</integer>
	<key>productID</key>
	<integer>1040</integer>
	<key>productName</key>
	<string>Griffin PowerMate</string>
	<key>productVers</key>
	<integer>1024</integer>
	<key>transport</key>
	<integer>1</integer>
	<key>usbLocationID</key>
	<integer>0</integer>
	<key>vendorID</key>
	<integer>1917</integer>
	<key>vendorIDSource</key>
	<integer>0</integer>
	<key>vendorName</key>
	<string>Griffin Technology, Inc.</string>
</dict>
</plist>
```

Im sure whatever the issue is, it's probably trivial but I don't know the more advanced capabilities of HID Remapper. The only thing I've used it for so far was to make a macro pad out of a USB Numpad, but that was easy.

jfedor2, 2025-02-18:
I have a theory on what might be happening here. It's more of a guess, but we can test it.

Try defining a custom usage like this (in the Settings tab):
[image]
Save, disconnect and reconnect HID Remapper and see if it makes any difference.

relink2013, 2025-02-18:
Alright! It definitely works! The scrolling is incredibly slow, but it's very smooth and consistent.

jfedor2, 2025-02-18:
Glad to hear. You can try to use the scaling parameter on a mapping to make it faster.

relink2013, 2025-02-24:
Everything is working perfectly. Thank you, I can finally just plug it in and use it without having to jump through hoops to set it up every time. 

I've hung onto this thing for years it's such a simple device but so incredibly useful. Plus it's really well built. I cant believe they stopped making them.