jfedor2/hid-remapper #214: Post expression monitoring?

MackNNations, 2025-02-20:
Is there a way to see the values that monitor results after the expressions are evaluated?    I know I can go to joypad.ai but that uses the the gamepad api in javascript and doesn't show the raw values form the hid remapper.    I'm still trying to sort out expressions for xbox controllers where the joy stick values range from -32768 to 32767 and they just don't add up.  I think the expressions page says that the values are converted to the range 0 to 255. Is that always the case?    Is there a way around the internal conversion ?

jfedor2, 2025-02-20:
The monitor shows the raw values coming from the plugged in device.

If you fetch the input state in the expression using `input_state` then you get the raw value. If you use `input_state_scaled` then you get the value rescaled to 0-255 range using the declared logical minimum and maximum.

You can use the `monitor` expression operation to send values to the monitor during expression evaluation.

At any point if you want to see what's on top of the stack you can do `dup 0xabcd0001 monitor` and the value will show up in the monitor as 0xabcd0001 (multiplied by 1000 I think). You can send multiple values by using different numbers, the 0xabcd0001 in this example doesn't mean anything, it's just made up.

MackNNations, 2025-02-22:
I tried dup-ing values and then `monitor`, but it does seem to show different values. 

I appreciate your help.  I know I have a lot of dumb questions.     I'll re-read the expression guide.