jfedor2/hid-remapper #218: Can't remap "V scroll" to "Ctrl + V scroll" (up scroll works, down scroll doesn't)

xuhdev, 2025-02-27:
I'm trying to create a mapping so that I can zoom in/out with mouse. I do this by observing that most browsers and other applications use "Ctrl + scrolling" to zoom in/out.

This is what I configured in the same layer:

|Input | Output|
--------|--------
|V Scroll | Left Control|
|V Scroll | V Scroll|

Then I switch to this layer, Scrolling Up works as expected just like Ctrl + Scrolling Up, but Scrolling Down doesn't work.

jfedor2, 2025-02-27:
Just create another V Scroll->Left Control mapping, but set the scaling parameter to "-1".

It works this way so that you can map scroll up and down to different keys. In your case you just want them mapped to the same thing.

xuhdev, 2025-02-27:
Awesome, it works! Do you have documentation anywhere? This looks like very hard to find out :)

jfedor2, 2025-02-28:
Yeah, I think this behavior was only ever discussed in this issue: #5 and I never updated the documentation.