jfedor2/hid-remapper #225: help with hold down macro/expression

tanadik, 2025-03-15:
hi! i was recommended here from reddit by jfedor.

i am having troubles with creating a function where i can toggle or hold a key being held down for a specified amount of seconds at a time and repeating that.

for example, if i tap or hold a key, it will hold they key down for 5 seconds, and release, and loop that continiously. i do not want it to turbo, just hold down or sticky.  

would such a thing be possible?

currently im using sticky function but it would be easier accessability to be able to loop it.

i have very limited programming knowledge and have no idea how to approach this, 

thank you so much

jfedor2, 2025-03-16:
What are you using this for? Is it important that the first interval is 5 seconds? Or would it be acceptable for the key to be released every 5 seconds, but for the period to not be synchronized to the keystroke that enabled it?

That could be done with a simple configuration like this:

[image]

Expression 1:
```
time 5000 mod 0 eq not
```

jfedor2, 2025-03-16:
For the first period to also always last 5 seconds, you can use a configuration like this:

[image]

Expression 1:
```
2 recall not /* previous layer state */
layer_state 0x02 bitwise_and
mul /* if layer 1 was just activated... */
0 /* ... reset counter */
1 recall /* otherwise fetch previous counter value */
ifte

1 add 5000 mod 1 store /* increment counter, wrap at 5 seconds */

layer_state 0x02 bitwise_and
2 store /* remember layer state for next frame */
```

tanadik, 2025-03-16:
working perfectly, thank you.
my wrist injury prevents me from using my left ctrl and shift, so i needed to find a work around tap, instead of holding down the buttons for gaming