jfedor2/hid-remapper #226: Question: Support for wired joysticks on iPadOS & iOS

techhitnz, 2025-03-15:
I have Retroarch on iPadOS 18.3, and I've tried getting my Taito Egret Mini Arcade stick working with it (without success).

I've tried mapping the Taito Egret to a Switch, PS4 Dualshock Controller, XAC, and Keyboard & Mouse.  But Retroarch doesn't detect any of those configurations

Retroarch detects my bluetooth keyboard without any issues.

iOS article says they officially support wired controllers, but they do not specify which ones:
https://support.apple.com/en-gb/guide/iphone/iph9d38dd45f/ios

News article says that iOS 18 adds wired Xbox controller support:
https://www.theverge.com/2024/10/7/24264620/iphones-now-support-wired-xbox-controllers
https://mastodon.social/@natbro/112621650100501936

Is this something that is possible to add support for?

jfedor2, 2025-03-16:
When in keyboard mode, does it work elsewhere on the iPad?

techhitnz, 2025-03-16:
No it does not.

jfedor2, 2025-03-18:
Check out the "analog stick cardinal directions to D-pad" and "analog stick cardinal directions to buttons" configurations in the Examples tab.

jfedor2, 2025-03-18:
And I'd check the gamepad modes again. If keyboard wasn't working for you before and now it is.

techhitnz, 2025-03-19:
> Check out the "analog stick cardinal directions to D-pad" and "analog stick cardinal directions to buttons" configurations in the Examples tab.

I've configured the controller as this, but only the buttons work and none of the joystick inputs on iOS in any application (such as Notes).  It does however, however, on a Windows PC - In normal windows environment (such as Notepad) and in Retroarch.

I think I'll give up on trying to get the workaround going by trying to get it functioning as a keyboard.  

I don't want to delve too much into the Retroarch configuration, but I have also tried modifying the input driver to SDL2, MFi and HID - None of which is able to get the controller working properly as a keyboard or gamepad.

Please consider this ticket a feature request to remap to MFi controller.  If this is something that's not on your roadmap, then feel free to close.  

Either way, I appreciate your time and effort on this project.