jfedor2/hid-remapper #230: Microsoft Xbox Adaptive Joystick is not a USB HID joystick

controllercustom, 2025-03-22:
The new Microsoft Xbox Adaptive Joystick does not work with hid-remapper because XAJ is not a USB HID joystick. The USB descriptors report the XAJ as a Microsoft vendor specific device. This was captured after upgrading to the latest firmware. This looks more like an Xbox gamepad than an ordinary joystick.

```
Bus 001 Device 019: ID 045e:0b1a Microsoft Corp. Xbox Adaptive Joystick
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass        71 
  bDeviceProtocol       208 
  bMaxPacketSize0        64
  idVendor           0x045e Microsoft Corp.
  idProduct          0x0b1a 
  bcdDevice            1.00
  iManufacturer           1 Microsoft
  iProduct                2 Xbox Adaptive Joystick
  iSerial                 3 xxxxxxxxxxxxxx
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0020
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass     71 
      bInterfaceProtocol    208 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               4
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               4
```

controllercustom, 2025-03-23:
I plugged it into an Adafruit Feather USB host board. Nothing shows up in the monitor tab. Perhaps it works with a 2 pico configuration.

jfedor2, 2025-03-23:
I tested it with the Feather, it works for me.