jfedor2/hid-remapper #233: XAC  and HID Remapper - How to Connect

Tsmith76, 2025-03-27:
Hi there!

Posted on your youtube comments, but figure here was probably a better spot to get my question answered. 

I'm struggling to get HID remapper to work with my Xbox adaptive controller. I bought the adafruit Rp2040, successfully connected it to my controller and remapped the inputs on my IFYOO gtp01 gamepad, and then connected it to my XAC - with no input detected. I disconnected it all, tried it with a Dualsense the same way you did in a previous video, and no dice there. I'm thinking it must be a problem with how I'm doing it on a hardware level - my RP2040 has an input slot for a USB-C and an input slot for USB-A, but not output dongles for either. So I'm plugging one adaptor cord from the controller into the USB-A slot and then another adaptor cord from the USB-C slot into the XAC. And I think something might be lost in that process, because the XAC doesn't detect that anything is there. 

Is there some other way I should be connecting the RP2040 to the XAC? Thank you! Sorry in advance for how utterly tech-incompetent I am.

jfedor2, 2025-03-29:
That is the correct way to connect it.

I don't have an explanation, but I've seen some weird behavior with the XAC where it would sometimes depend on the USB cable that I used between the XAC and the PC for example (a long cable didn't work, a short one did). Might be some power or voltage drop issue, I'm not sure.

The XAC can also take external power over a barrel connector, but I haven't tried that. Might help, but you'd need an additional power supply.

Tsmith76, 2025-03-30:
I'm not sure if I'm doing anything right, to be honest. I've got my controller mapped in the HID Remapper screen - when I bring up Joypad.ai, what I'd expect to happen is to see my mappings mirrored to the gamepad on-screen, how I would want them. I've mapped my touch pad controller to the right analog stick, because that's how I want to use it - also flipped, so the controller can sit next to me on the armrest more easily. 

I've attached a copy of all of my settings. Notice on joypad.ai how, when I'm pressing the 'C' button which should be mapped to 'X' button on XAC, it's instead detecting the analog stick for some reason? 

[image]
[image]
[image]

jfedor2, 2025-03-30:
When in "XAC/Flex compatible" mode, the inputs won't match on browser based gamepad testers.

Tsmith76, 2025-03-30:
But my settings otherwise look correct, and theoretically it should be connecting to the XAC?

jfedor2, 2025-03-30:
Yes. As I said try shorter USB cables or plugging other devices into HID Remapper to see if anything works.

Tsmith76, 2025-03-30:
How short would you recommend? I've got one cord that's one foot and another that is three feet, so I'll probably need to buy more if I need shorter.

jfedor2, 2025-03-31:
That's a red herring. It doesn't matter if it works with the XAC directly, after it goes through HID Remapper it all looks the same.

If a device works with HID Remapper while it's plugged into a PC, but doesn't work when HID Remapper is plugged into the XAC then it's some weird power issue that I don't fully understand, but one thing I've noticed is a shorter USB cable seems to help (with the DualSense for example).

The Switch Pro controller doesn't work with HID Remapper at all because it requires some proprietary initialization to work when plugged into anything.

Tsmith76, 2025-03-31:
I'm a little confused though, because why would one device work when going through HID remapper but not the others because of the cable? Is it something to do with the controller pulling different amounts of power - like the Joystick was just more efficient or something? 

If so, I'll give one last try at buying a short usb-a to usb-a cord to connect HID Remapper to the XAC, something like 6 inches or something, and hope for the best. Otherwise, I'll just have to guess that there's something wrong with my HID remapper, my XAC, or both. 

Does the length of cable going from HID remapper to the seperate controller itself matter, or just the cable going from the remapper to the XAC?

jfedor2, 2025-03-31:
In my tests I saw a difference when I changed the cable between the PC and the XAC, but it's probably the total length of all cables between the PC/console and the device that's plugged into HID Remapper.

If you can get it to work with shorter cables then look into getting an external power supply for the XAC.