jfedor2/hid-remapper #237: Extra Mouse Button not Recognized w/ 2.4 Wireless Elecom EX-6

sternskr, 2025-03-31:
I suspect this is pretty similar to the other elecom issues in the past where some HW buttons were not detected for some Elecom mice. Not sure what you might need from me to help figure it out/confirm it is an issue but I'd love to provide help if I can

jfedor2, 2025-03-31:
It's likely the same issue that can be fixed with custom usages.

You could try dumping the report descriptor and some traffic where you press the buttons that don't work.

sternskr, 2025-04-01:
I’m assuming it’s similar to that thread and I dump via a USB wireshark capture. I’ll give that a shot, I’ve never done a USB capture but I’ve used wireshark so I should be able to figure it out, I’ll do that once I’m home