jfedor2/hid-remapper #238: Prefix all keypresses with a custom key

oscarorange, 2025-04-04:
I would like for all my key presses to be prefixed with `F24 down` and suffixed by `F24 up`, if I press multiple keys at the same time I want them to be sandwiched in between instead of pressing and releasing `F24` multiple times. For example if I click `A` I would like for it to press `F24 down` > `A down` > wait for `A` release > `A up` > `F24 up`, if I press A and B at the same time I would like it to press `F24 down` > `A down` > `B down` > wait for `A` or `B` release > `A up` > `B up` > `F24 up`. It would be good if I can change `F24` to something else in the future easily too.

I'm sure this is possible but I'm struggling to figure it out, please may someone help me with the expression?

jfedor2, 2025-04-11:
We don't really have a way to check if any key is pressed so this is going to be tedious regardless of how we decide to implement it, because we're going to have to list all the keys.

One way to do it is disable unmapped inputs passthrough on layer 0:

[image]

Create an expression like this:
```
1 recall
2 store /* any key pressed previous frame */

0x00070004 input_state_binary
0x00070005 input_state_binary bitwise_or
0x00070006 input_state_binary bitwise_or
/* ... */
1 store /* any key pressed */

1 recall
2 recall bitwise_or
3 store /* we want F24 pressed */

1 recall
4 store /* we want all keys to be passed through */
```
And mappings like this:

[image]

The way it works is F24 will be pressed whenever a key is pressed or was pressed on the previous frame, giving us the sandwich.

Normal keys only work on layer 1 and we activate layer 1 when register 4 is set to 1. (Because the layer activation is processed first, before expressions are evaluated, it effectively uses the register state from the previous frame so we don't have to check if any key was pressed on the previous frame - this gives us the 1 frame delay automatically.)

The only issue is that this expression only works for the A, B and C keys. To make it work for the other keys they have to be explicitly listed (where the `/* ... */` is).

oscarorange, 2025-04-11:
Thanks a bunch ❤️ This seems to work as expected.

I added all the hex numbers from [`usages.js["source"]`](https://github.com/jfedor2/hid-remapper/blob/06f0acd6b3a8c76461f3d39277a7aa06bed81bed/config-tool-web/usages.js#L58-L190).

Please may you let me know if it would also be possible to have multiple sandwich prefix and suffix keys? If I wanted all my key presses to be prefixed with `Right Control` and `Right Shift`, or `Right Control` and `F24` for example?

Inputting `A` would output `Right Control down > Right Shift down > A down > wait for A release > A up > Right Shift up > Right Control up >`.
Inputting  `A and B` should output  `Right Control down > Right Shift down > A down > B down > wait for A release > A up > wait for B release > B up > Right Shift up > Right Control up`.

jfedor2, 2025-04-11:
If it's okay for all the sandwich keys to be pressed and released simultaneously then you can just make another mapping with Register 3 as the input and whatever key you need as the output.

oscarorange, 2025-04-11:
Awesome! Thanks.