jfedor2/hid-remapper #239: Weird scrolling behavior, also feature request for BOOTSEL button

chalermla24, 2025-04-05:
Using 2-pico version, Sensei Ten mouse (This happens with any mouse).  
Randomly the scrolling speed will go super slowly, like when setting the "lines" in Control Panel to a low amount.
Can force this to happen by either holding the BOOTSEL button for some time and letting go on the B side, or re-plugging in the mouse on the B side without disconnecting the A side from the PC.
Current workaround for the problem is to re-plugging in the A side or reset the USB port.
Also feature request. Can you make the BOOTSEL button reset the device? Thanks.

scatterthought, 2025-04-07:
I've noticed the same thing with the single-Pico version. When it happens randomly, I think that's from the Pico losing power for a split second and rebooting.

I had an HID Remapper on a powered USB hub. When I'd get a static shock from touching other devices plugged into that hub (a Seenda trackpad and a fingerprint reader), the slow-scrolling behaviour would start and I'd have to re-plug the HID Remapper. Similarly, plugging/unplugging devices would sometimes cause it as well.

I would guess that the slow scrolling comes from the Pico being powered before the input device, and misinterpreting the polling speed. It sounds similar to the past issues with USB hubs being plugged into HID Remapper dongles.

I've also thought about the BOOTSEL button being used to restart the dongle. On my Waveshare RPi-Zero I have separate BOOT and RESET buttons.

jfedor2, 2025-04-10:
Sounds similar to #76 (though there may be two separate issues, one causing very slow scrolling and one causing very fast scrolling). I don't have a solution, but my hypothesis is still that the OS thinks we're in high resolution scrolling mode and HID Remapper doesn't or the other way round. Hard to imagine HID Remapper restarting without the OS noticing so I'm not sure how we're getting to that state.

chalermla24, 2025-04-10:
Maybe it's that the HID Remapper didn't actually "restart" to the OS but itself does restart (maybe by losing but then getting power very fast). Or it could be that the devices connected to the HID Remapper actually restart?

jfedor2, 2025-04-10:
Yes, something like that, but I don't know how it's possible, USB isn't stateless, devices are enumerated, get assigned addresses etc.

(Of course it's also possible that the fault lies with the OS, but it's probably best to consider other explanations first.)

scatterthought, 2025-04-10:
It does sound similar to #76, but I never have the issue after my computer goes to sleep. I wonder if that's because I've always had it plugged into a KVM hub that always shuts down power to USB devices. My HID Remapper has to boot every time.

I just found your old [comment on QMK about the hi-res scrolling](https://github.com/qmk/qmk_firmware/issues/17585#issuecomment-1238023671). I'm mentioning it here as it helped me understand your earlier comments a little more (assuming it's all still valid).

Would you be able to add a toggle to the web interface so that we can enable/disable the reset on high-res scrolling? I see the comment that the problem came back for someone, but I feel like we need to test that more.

scatterthought, 2025-04-11:
Just wanted to share my workaround for managing the scrolling issue without having to unplug the dongle.

If you have a scroll wheel on your device:

1. Make all of your base mappings (that should work all of the time) available on both Layer 0 and Layer 1.
2. Change the scrolling scale on Layer 1 to be 120x the default scrolling behaviour
3. Add a button mapping that turns on a sticky Layer 1. I use a button combo that I'm unlikely to press accidentally. 

When the dongle glitches, use the sticky button mapping to change the device to Layer 1. When the dongle is rebooted, it will default back to Layer 0.

If you use a trackball with hold-to-scroll mappings (as I do), it gets a little more complicated due to the additional layers. I still put the base mappings on Layer 0 and Layer 1, but the default scroll scaling moves to Layer 2 and the 120x scroll scaling is on Layer 3.

scatterthought, 2025-04-16:
I'm finding that after I've enabled my Layer 1 workaround for scrolling, HID Remapper will revert to Layer 0 after some time has passed (even though I'm not disengaging the sticky Layer 1). This leads me to think that the dongle is resetting itself frequently without reporting to the OS, and not because of static discharges (as I previously thought).