jfedor2/hid-remapper #243: Remapping U and Shift-U to different Macros

dhuffer, 2025-04-09:
I'm trying to remap U and Shift-U to different Macros and can't find a way.

I have a layer that is activated by the CAPS key:

[image]

Now I want to map a different macro to the same U key, but this time when it is pressed with shift.

jfedor2, 2025-04-10:
You can either make Shift activate another layer or make an expression like this and map Register 1/2 to the macros you want.
```
0x000700e1 input_state_binary not /* shift not pressed */
0x00070018 input_state binary /* U pressed */
mul
1 store
0x000700e1 input_state_binary /* shift pressed */
0x00070018 input_state binary /* U pressed */
mul
2 store
```

dhuffer, 2025-04-13:
Many thanks, that worked 😀

closed 2025-04-13 by dhuffer