jfedor2/hid-remapper #244: Rev 8 - GPIO buttons not registering

Bonktacular, 2025-04-12:
Got my Rev 8 board back from JLPCB, flashed the firmware you compiled, slapped on my buttons and... Only GPIO buttons 1-6 register inputs. 

Tried several of the boards, all the same results. Was there a GPIO extender missing from the firmware? I'm pretty new to hardware/firmware. 🤔

jfedor2, 2025-04-12:
How did you mount the switches, can you show a picture?

Does it work if you short the pads with tweezers or a piece of wire?

Bonktacular, 2025-04-12:
I can show a picture when I get back later this evening, thank you for the response. the kailh browns are soldered in by their contact pins, and shorting the contacts does not produce anything on the monitor.

Bonktacular, 2025-04-13:
Evidently my soldering job was garbage, I heated the pins again and they're beginning to show up. Sorry to bother you!

closed 2025-04-13 by Bonktacular