jfedor2/hid-remapper #245: LED interaction

RuloH, 2025-04-20:
Hi. Is there a way to make a neopixel led such as the one integrated in the zero Pico to blink in different colours depending on momentary switches activity hooked ofc in different gpios?

Your code is amazing. Thanks again for such remarkable work.

jfedor2, 2025-04-20:
Regular LEDs yes, NeoPixels no.

MarqueIV, 2025-05-30:
Hi RuloH.  I've been thinking of doing the very same thing.  As stated above, you can't with the code as-is, but there are two ways I can think of that may make this possible. Unfortunately, both require programming and compiling on your end.

The first is to simply use a separate board to drive the neopixels, then trigger what you want via standard GPIO.  That requires no changes to HID-Remapper's firmware, but does require you to write your own on a separate board, then wire them together.

The other way is to mod the HID-Remapper source yourself to add a neopixel driver in one of the programmable interrupts on the RP2040 (Adafruit has good examples you can crib from), then possibly add your own expressions to run your custom LED code, or create your own outputs (like the GPIO stuff) to do that.  The advantage of this approach is it requires no new hardware.  The disadvantage is you'd then be responsible for maintaining your own fork, and merging down new changes.

I'm personally trying the latter as I already have experience adding new commands to the expression language, and I like that it's then all in one board.  However, if things get too hairy, falling back to the GIO-as-triggers approach will more than suffice with a small sister RP2040 board from XIAO as an example.

Hope this helps!