jfedor2/hid-remapper #246: Question about remapping contour Shuttle v2 Outer Wheel

Zedibee, 2025-04-21:
Hello,

I had a Contour Shuttle Pro v2 device that I'm essentially trying to make on-board memory out of. The previous discussion #207 was very helpful in getting the previous buttons mapped correctly; However, I was wondering if some more advanced functionality might be possible.

Specifically, the outer jog wheel which acts as a VScroll output has values ranging from -7 to 7. I was wondering if it might be possible to map each of these values to a different function, such as -7 to Alt+7, and +7 to Ctrl+Alt+7, with similar remappings for the other values. I tried some manual JSON edits with the use of LLM assistance but to no avail. Any idea if such functionality is possible through expressions or some other way? Thanks again for all your help and wish you the best.

Zedibee

[image]

jfedor2, 2025-04-21:
Assuming you're using the custom usages from #207, you can make an expression like this:
```
0xffff0038 input_state 1 eq 1 store
0xffff0038 input_state 2 eq 2 store
0xffff0038 input_state 3 eq 3 store
0xffff0038 input_state 4 eq 4 store
0xffff0038 input_state 5 eq 5 store
0xffff0038 input_state 6 eq 6 store
0xffff0038 input_state 7 eq 7 store
0xffff0038 input_state -1 eq 8 store
0xffff0038 input_state -2 eq 9 store
0xffff0038 input_state -3 eq 10 store
0xffff0038 input_state -4 eq 11 store
0xffff0038 input_state -5 eq 12 store
0xffff0038 input_state -6 eq 13 store
0xffff0038 input_state -7 eq 14 store
```
And then create mappings with Register 1-14 as input and whatever you want as output.

If you want to map to something like Ctrl-Alt-7, just create three separate mappings with the same input, one with output set to Control, one with output set to Alt, and one with output set to 7. You can also use macros, but for simple key combinations like this it's not necessary.

jfedor2, 2025-04-22:
Macros no, but if you make regular mappings as mentioned above then that's how it should behave already.