jfedor2/hid-remapper #247: Vivaldi browser fails to open HID device - Use Chrome instead

akseidel, 2025-04-25:
The Vivaldi browser currently fails to open HID devices. Therefore the HID Remapper Configuration website will not work in a Vivaldi browser. It is a known bug at Vivaldi that has been known for some time but that is not being addressed. The issue applies to multiple operating systems.

The error you get trying to connect to the HID Remapper is the "NotAllowedError: Failed to open the device.
If you're on Linux, you might need to give yourself permissions to the appropriate /dev/hidraw* device."

On OSX, the HID Remapper Configuration website functions in a Chrome browser.

valdotdev, 2025-05-24:
There's a year old thread in the Vivaldi forums regarding the issue. They're aware but for now the workaround is using a PWA in Vivaldi instead. Right now it works on all browsers and OSs except Vivaldi so I doubt it's an issue with HID remapper.

https://forum.vivaldi.net/topic/97917/hid-devices-issue?lang=en-US&page=1

boxcat-chris, 2025-08-07:
Just ran into this same issue in Chromium on a Raspberry Pi 5.

boxcat-chris, 2025-08-07:
Yes, I got the same hidraw permissions message reported by the person that opened this issue. I went to Google it and promptly fell down a rabbit hole of many different answers and everybody telling each other they’re wrong. Being relatively new to using Linux as a desktop OS, I ended up more confused than enlightened.

There are people saying to add a new udev rule and then add yourself to that, others saying that’s a security issue, and other people suggesting temporarily granting blanket read/write permissions, doing whatever you need, then revoking them afterwards, et cetera and ad nauseam. I don’t know which approach is right.

jfedor2, 2025-08-09:
Sorry, perhaps I wasn't clear. This issue is about the Vivaldi's browser failure to implement WebHID properly. The symptoms are the same, but you likely just need to give yourself permissions to the device, like the error message says. What you read on the Internet is true, you can do it one time with chmod or chown or you can create an udev rule for it to happen automatically every time.

The udev rule might look like this:
```
SUBSYSTEMS=="usb", ATTRS{bInterfaceNumber}=="?*", ENV{ID_USB_INTERFACE_NUM}="$attr{bInterfaceNumber}"
KERNEL=="hidraw*", SUBSYSTEM=="hidraw", ATTRS{idVendor}=="cafe", ATTRS{idProduct}=="baf2", ENV{ID_USB_INTERFACE_NUM}=="01", TAG+="uaccess"
```

(This assumes the default keyboard/mouse mode, other modes have their own vendor/product IDs so you'd have to adjust those.)

If you just want to do a chmod or chown then you only need to figure out the right device number. HID Remapper will have two hidraw devices which you can see in dmesg, you want the second one, which is the configuration interface.

The udev rule will only grant access to the currently logged in user (physically at the computer, not via ssh).

boxcat-chris, 2025-08-09:
Oh. I understand now. Thank you, I’ll take a poke at that, and I appreciate your help!