jfedor2/hid-remapper #248: USB Hub with v3 Board?

bsoben, 2025-04-25:
I have been trying to use a small USB hub to hook up both a keyboard and mouse to a v3 board and I'm never able to get anything to show up.  Am I missing something?

>What you can plug into HID Remapper
>
>Wireless receivers are supported and multiple devices can be connected at the same time using a USB hub (with different mappings for each device if desired).

jfedor2, 2025-04-25:
Try a different hub, not all of them work unfortunately.

bsoben, 2025-04-26:
I tried 3 different hubs, none worked, do you have a specific model that you have been successful with??

jfedor2, 2025-04-26:
Hubs with the FE1.1s chip should work, for example: https://www.waveshare.com/usb-hub-hat-b.htm

Also CH334 I think.

For the dual Pico variant also the SL2.1a chip works, it's what the variants with a built-in hub use.

jfedor2, 2025-04-28:
Should work the same. For some hubs the order of plugging things in matters (plug into HID Remapper, devices into hub, HID Remapper into the PC).

jfedor2, 2025-04-28:
Or maybe some recent changes in TinyUSB improved things - the version used in HID Remapper lags a bit. Need to test.

jfedor2, 2025-04-28:
Here's a test build with TinyUSB updated to the latest version. Feel free to try it and see if it affects hub compatibility.

https://github.com/jfedor2/hid-remapper/actions/runs/14713179580

("artifact" link at the bottom)