jfedor2/hid-remapper #249: web tool in archive — Incompatible version

Famicombro, 2025-04-26:
I updated the firmware to the latest version, downloaded the archive from github and tried to use the local version of the web configurator from the web-tool folder, I use spark as a server. But when I try to connect to the device, it gives an error «Incompatible version. Please consider upgrading your HID Remapper firmware to the latest version».

jfedor2, 2025-04-26:
Code in the repository is the same as what's on https://www.remapper.org/config/

Does that give you the same error message? If so then you really have old firmware.

Famicombro, 2025-04-26:
Maybe I'm doing something wrong. now I have another problem, the device connects, but then it doesn't work. There may be some problems with my server.

[image]

jfedor2, 2025-04-26:
Does it happen when you use the website at https://www.remapper.org/config/ ?

Famicombro, 2025-04-26:
when I use remapper.org/config, everything works fine. But I like having a local version. It used to work well, but in recent versions some strange things are happening.

jfedor2, 2025-04-26:
I understand, I'm asking because it should work the same, there's nothing special about that website.

Maybe your browser is caching old code? Tried shift-reload?

Famicombro, 2025-04-27:
I fixed the issue. For some reason, the archive downloaded from github contained broken files. I replaced them all again and everything worked. I'm sorry for the stupid ticket, I should have been more careful.