jfedor2/hid-remapper #250: 3D Connexion SpacePilot remapping

alextrical, 2025-04-26:
Hi All

I was wondering what would be required to remap a 3D Connection SpacePilot.
3D Connection has long EOL'ed this product so thaty can be got for under $50, however the driver support is showing issues (they want people to keep buying the latest hardware)
It would be great to be able to re-map this to a 6 axis HID device that can be used by Fusion and FreeCAD on Mac or Linux

jfedor2, 2025-04-26:
HID Remapper can't currently act as a SpaceMouse so you wouldn't be able to use it in Fusion etc.

alextrical, 2025-04-27:
I like the word "Currently"
I do wonder how much would be involved in making the output look like a more modern space mouse (PID) I have no idea what has changed with the driver protocol's with newer devices, and part of me hopes that the main change is the PID, and that the products share the same protocol.

Time will tell, but any progress I can help with would be awesome.

Even if the resulting translation is only compatible with the open source space mouse driver, I would class it as a win

jfedor2, 2025-04-28:
The data format is very simple, there are just some quirks that don't fit well within HID Remapper's way of doing things and would require some special casing.

A SpaceMouse always sends the translation/rotation reports in pairs and some usages are flagged as relative in the descriptor when they should be absolute (or maybe it's the other way round, it's been some time since I've looked at this).

If you're only interested in making an older unsupported device look like a modern SpaceMouse and don't care about the rest of the remapping capabilities, the firmware for that would be relatively simple. Could probably hack something together by taking the relevant parts from these projects:

https://github.com/jfedor2/wheel-adapter
https://github.com/jfedor2/magellan-spacemouse
https://github.com/jfedor2/spaceball-2003

alextrical, 2025-05-03:
Good day

I've tried using the Spaceball and Magellan firmware to simulate a space mouse pro, I'm assuming it should expose itself as a Space mouse Pro regardless of a serial device being present of not, but I am having difficulty getting the 3D connexion driver to detect the device.

I'm using macOS 16, on Intel iMac and a M1 MacBook pro, but neither detect the device in the driver configuration.

Is there anything I can do to debug why this may be happening?
Should the RP2040 expose itself as a Space Mouse Pro regardless of a serial device being attached?

jfedor2, 2025-05-05:
I forgot about that. :) I have the source code somewhere, but that was a hack on top of an old version of HID Remapper so I don't know if it's a good starting point. I have a more cleaned up version that I can probably put up if you want to play around. I wasn't fully satisfied with it, but I don't remember the details.

alextrical, 2025-05-06:
if you could make it available, that would be amazing.
I think the goal would be getting something added back into hid-remapper for the maker community, and could even alow remapping of a flight simulator stick. Seeing as the SpaceMouse is a pricy piece of tech, but pretty neat to get an old device working again with the new driver. I think you can see there is some demand from the comunity wanting this, by virtue of yourself making the serial adapter and others asking for USB re-mapping too ;)

Side question:
From what I've seen in the other threads, it looks like the RP2040 will have to be setup in Dual mode to support the "Space Mouse Classic USB", Im curious about the Dual RP2040 mode, is it that the RP2040's USB PHY peripheral is needed to communicate with tighter timings than the Bit Banged version? 
If so would it be possible to Flip the ports on the RPi, so that the hardware PHY is connected to the USB HID device, but the Software/BitBanged port runs in device mode to connect to the PC instead?

braxlan, 2025-05-06:
> [...] If so would it be possible to Flip the ports on the RPi, so that the hardware PHY is connected to the USB HID device, but the Software/BitBanged port runs in device mode to connect to the PC instead?

See [issue 222](https://github.com/jfedor2/hid-remapper/issues/222)

jfedor2, 2025-05-06:
> if you could make it available, that would be amazing. I think the goal would be getting something added back into hid-remapper for the maker community, and could even alow remapping of a flight [...]

Yes, that is certainly the goal.

> Side question: From what I've seen in the other threads, it looks like the RP2040 will have to be setup in Dual mode to support the "Space Mouse Classic USB", Im curious about the Dual RP2040 mode, [...]

As mentioned it should be possible in theory, but I haven't been able to make it work. But why do you ask, do you have reasons to believe your SpacePilot will have issues with the single Pico variant using a bit-banged host interface?

jfedor2, 2025-05-06:
Here's what I have now:

https://github.com/jfedor2/hid-remapper/tree/spacemouse-our-descriptor

Compiled firmware files here ("artifact" link at the bottom):

https://github.com/jfedor2/hid-remapper/actions/runs/14865664049

It seems to work fine when I plug in a SpaceBall 5000 USB or a SpaceMouse Compact (which doesn't require an adapter, but just for testing).

I haven't tested on a Mac, but I set the vendor string to "3Dconnexion".

You will need to host a copy of the configuration interface (`config-tool-web` folder) somewhere (can be local with something like `python3 -m http.server`, but won't work with `file:///...`).

Then just set the emulated device type. Unmapped inputs passthrough should work for spacemouse-type devices, but you could of course try and map something like a game controller too (as always keep 3Dconnexion's software license in mind).

alextrical, 2025-05-06:
@jfedor2 You're an absolute wizzard, it works PERFECTLY with the SpacePilot, to allow it to function on the Mac.

I've not been able to confirm that buttons all work, though a large chunk of them seem to in Fusion360. (I will play around with mapping them over the coming week)

The 6-DoF mouse has all of the axis' mapped correclty, and can run through the test games really well. One of them has an issue but that looks to be the porting of the fame to Arm, not the input.

Amazing job.

Is there anything I can do to progress this, to debug, help test or develop, to make it suitable to be part of the main branch for others to make use of too?

alextrical, 2025-05-06:
> As mentioned it should be possible in theory, but I haven't been able to make it work. But why do you ask, do you have reasons to believe your SpacePilot will have issues with the single Pico variant using a bit-banged host interface?

I assumed from seeing the other user with a "Space Mouse Classic USB" that it wouldnt work, so I went strait for the Dual RP2040 setup, and wondered if having the RPi port for the USB peripheral would solve the issue for all devices. It was more out of curiosity than anything else, but I could poke around and see if i can get an example of Bit Banged USB Device working as a PoC

I will get a single RP2040 setup tomorrow to test the single device firmware ^_^
Woop, looking forward to using a Space Mouse for CAD work tomorrow

alextrical, 2025-05-07:
Likeley outside of the scope of this Issue, but notes of the issues im seeing:
Once the Hid-Remapper is configured to be an Emulated device type of "SpaceMouse Pro", and the RP2040 power cycled, the WebUI can nolonger configure the device. Flashing [flash_nuke.uf2](https://datasheets.raspberrypi.com/soft/flash_nuke.uf2) followed by the Hid-Remapper FW is the only way ive found so far to get the config to work again>

The SpaceMouse Pro doesn't have enough button inputs to account for the available inputs of the SpacePilot, I will see if i can compile the FW with the VID and PID for SpaceMouse Enterprise

Mapping notes:
[image]
Physical Button    |Usage           |Usage Name
1                            |0x00090001|Left button
2                           |0x00090002|Right button
3                           |0x00090003|Middle button
4                           |0x00090004|Back
5                           |0x00090005|Forward
6                           |0x00090006|Button 6
T                           |0x00090007|Button 7
L                           |0x00090008|Button 8
R                           |0x00090009|Button 9
F                           |0x0009000a|Button 10
ESC                      |0x0009000b|Button 11
ALT                       |0x0009000c|Button 12
SHIFT                   |0x0009000d|Button 13
CTRL                    |0x0009000e|Button 14
Fit                         |0x0009000f|Button 15
Panel                    |0x00090010|Button 16
Sensitivity Up       |0x00090011|Button 17
Sensitivity Down  |0x00090012|Button 18
DOM                     |0x00090013|Button 19
3D Lock               |0x00090014|Button 20
Config                  |0x00090015|Button 21

SpaceMouse Pro
Physical Button    |Usage           |Usage Name
MENU                   |0x00090001|Left button
FIT                        |0x00090002|Right button
T                           |0x00090003|Middle button
R                           |0x00090005|Forward
F                           |0x00090006|Button 6
Roll CW                |0x00090009|Button 9
1                           |0x0009000d|Button 13
2                           |0x0009000e|Button 14
3                           |0x0009000f|Button 15
4                           |0x00090010|Button 16
ESC                      |
CTRL                    |
ALT                       |
SHIFT                   |
3D lock                 |

alextrical, 2025-05-07:
Mappings all sorted, and placed in the Examples.JS file.
All buttons bar 5,6, Panel, -,+ and Dom are mapped and working.

The only outstanding bug ima aware of, is the need to wide the flash before being able to connect to reconfigure the device. Is that because of the Change in VID&PID? 

[image]

alextrical, 2025-05-10:
Its not the VID & PID being changed that causes the issue. There is something that is preventing the Configurator from connecting to the device when configured as a Spacemouse.

I will continue to investigate what may be the cause of the issue.

MarqueIV, 2025-05-30:
Been following along with this for a while.  Love where this is all going. I actually modded my SpaceMouse Enterprise to have a USBC port on it rather than that stupidly-long USB-A cable.  Game changer!  I have about three other variants of the device too.  Love these things.

Couple questions regarding your work since you have yours already running successfully.

1. From the perspective of mapping inputs, do the six axes work the same as a joystick's axes (i.e. can you make them control scrolling, panning, volume, etc. in regular software/OS stuff?)

2. Does this mean you can do the reverse... make a joystick, gamepad, keyboard, etc. appear as a SpaceMouse to the 3DX software?

3. Do you have any way to control what's on the device's screen itself, or is that all/only from the 3DX driver itself?

Thanks, and can't wait to experiment with your work here!  Keep up the great work!