jfedor2/hid-remapper #251: Mouse operation not supported under Chrome OS

fakemanhk, 2025-04-28:
Hi,

I found this an excellent project because most software mappers only available to Mac/Windows, but I use Linux/Chrome OS a lot.

I purchased Waveshare RP2350A USB-A and updated firmware, tested with most simple "mouse left/right key swap" which works nice under my Linux laptop. However when I move the whole set to my Chromebook, although I still see 2 HID devices being recognized like Linux, only "keyboard" can be recognized in Chrome OS setting, which means I can only use HID remapper with keyboard only, so how should I troubleshoot this?

braxlan, 2025-04-28:
Hi,

it's just a wild guess, but it might be the case that Chrome OS only supports devices with "HID Boot Protocol", which is not the case for the mouse part of the hid-remapper, see [issue 94](https://github.com/jfedor2/hid-remapper/issues/94).

fakemanhk, 2025-04-29:
> It works for me. The only thing that doesn't work on ChromeOS as far as I
> know is absolute mouse.

Does it show up as single or multiple device?

jfedor2, 2025-04-29:
Not sure what you mean. It shows up under keyboards and it shows up under mice, like any combo device.

fakemanhk, 2025-05-02:
Because in my Chromebook it's only showing extra keyboard but no mouse,
looks like it never recognize any mouse device.

fakemanhk, 2025-05-12:
> It works for me. The only thing that doesn't work on ChromeOS as far as I know is absolute mouse.

May I ask if you are using single or dual Pico? If dual Pico is the only way then I'll go for it.

jfedor2, 2025-05-12:
Both should work. There's no difference between them on the computer side.

fakemanhk, 2025-05-16:
OK I found the problem (or is it expected?), when I plug the Pico to Chromebook without mouse on the other side of Pico, it won't work in ChromeOS; When I plug the mouse to Pico and plug the USB-C to Chromebook, the mouse will work.