jfedor2/hid-remapper #255: incompatible keyboard: Sboc keeb / Surmen v500

raould, 2025-05-07:
hello,

(Thank you for HID-Remapper, I have a Feather RP2040 USB Host board from Adafruit, and I put the firmware on it yesterday.)

The incompatible keyboard is labeled "Surmen v500" but was sold as "Sboc Keeb" and it has a very good feel so I would like to use it if safe to do so. Keystrokes do not get through HID-Remapper. Wondering if the keyboard is "non standard" or "bad" in some way? Since I did reproduce this problem on 2 of 2 such keyboards I had available, I assume it is not a QA fail on the keyboard but instead something consistent about the device's use of USB protocol. (All my other keyboards tested thus far do work well and show keystrokes through HID-Remapper.)

https://www.amazon.com/SbocKeeb-Mechanical-Keyboard-Swappable-Pre-lubed/dp/B0D876CRS1?th=1
https://surmenhub.com/products/surmen-75-percent-gaming-keyboard-hot-swappable-with-side-light-strip-82-keys-wired-mechanical-keyboard-with-knob-arrow-keys-rgb-backlight-for-win-mac-82-sakura

Windows system report of the keyboards: I believe the first 2 are this keyboard, the last is the laptop's own keyboard.
VID 258, PID 0049
[device_dump.txt](https://github.com/user-attachments/files/20073927/device_dump.txt)

My attempt to get USB log for debugging:
[sboc_keeb_surmen_v500.pcapng.gz](https://github.com/user-attachments/files/20073929/sboc_keeb_surmen_v500.pcapng.gz)

Thank you for any insights.

jfedor2, 2025-05-09:
Hard to say, some devices just don't work.

You could try a dual Pico variant or the test build from here ("artifact" link at the bottom) that uses a newer version of the TinyUSB library: https://github.com/jfedor2/hid-remapper/actions/runs/14713179580

raould, 2025-05-09:
Thanks for the info, I will try out the newer version.

closed 2025-05-09 by raould

raould, 2025-05-26:
for posterity: the TinyUSB build *does* work for me on the Ada Fruit RP2040 feather. Thank you!