jfedor2/hid-remapper #256: Does it support rp2350A (pico 2w)

kaustuv90, 2025-05-07:
I know it support rp2350 but does it support rp2350A ?

jfedor2, 2025-05-07:
All the builds we have are for the RP2350A.

(There's no vanilla "RP2350", there's RP2350A and there's RP2350B.)

kaustuv90, 2025-05-07:
So, as I am using Pico Pi 2W, it has wifi, so we can host the webpage to change the config in the board itself and access with local IP to modify the configs. In that way, we don't need to connect to the PC to change the config. to be clear, wanna use for PS5 to be able to use keyboard and mouse.
I know we can set keyboard shortcuts to change between preset layers but a locally hosted webpage to modify the config will be cool to have, I think.