jfedor2/hid-remapper #257: Expression for deadzone and input scaling - XAC + Joystick

dblaze025, 2025-05-08:
I'm using two joysticks, both with HID Remappers plugged in to an XAC to run a HOSAS setup. 

The game I'm playing has a large deadzone built into the left stick. You have to move the stick around 60%-75% of it's travel from center to get a response in the game. This is no fault of the HID Remapper/XAC, it behaves the same way with a regular Xbox controller. 

I'm wondering if it would be possible to write an expression that would apply a scaling function to the input to overcome this. 

A graphical representation of the function would look something like:

[image]

I was able to get a basic deadzone only function working with a lot of guidance from ChatGPT:

`0x00010030 input_state_scaled dup -128 add abs 4 gt swap 128 ifte`

But to be honest I don't understand RPN well enough to implement the rest. Can you help please?

jfedor2, 2025-05-09:
For a per-axis deadzone/scaling you could use an expression like this (and same of the other axis):
```
0x00010030 input_state_scaled
128 sub
dup sign swap
abs dup 4 gt swap
4 sub
50 mul
124 div
77 add
mul
mul
128 add
```
Depending on how the in-game deadzone works, radial deadzone/scaling might be better:
```
0x00010030 input_state_scaled 128 sub dup mul
0x00010031 input_state_scaled 128 sub dup mul
add
sqrt
dup 4 gt swap
4 sub
50 mul
124 div
77 add
mul
/* vector magnitude */
3 store 

0x00010030 input_state_scaled 128 sub
0x00010031 input_state_scaled 128 sub
atan2
/* vector angle */
dup
sin 3 recall mul 128 add 1 store
cos 3 recall mul 128 add 2 store
```
(And map register 1 to stick X and register 2 to stick Y.)

dblaze025, 2025-05-10:
Haven't dove in to the radial deadzone version, but the single-axis method seems to work as intended, thank you! 

The logic behind this stuff is really cool once you understand it. Like how `dup sign swap` essentially decides which direction you're moving the stick by pushing +1/-1/0 then stores that on the bottom of the stack to be used later in the calculation. Simple in execution but the way its done is very clever. 

As far as adjusting it goes, am I understanding these values correctly?

`4` deadzone, pretty straightforward

`124` input range span (center - deadzone = 128 - 4 = 124)

`77`  deadzone jump 

`50` = slope of the scaling line after the jump, the higher this number, the faster output reaches maximum. not slope in a traditional y=mx+b sense, but it does determine the angle of the scaling line. to ensure 0=0 and 255=255, we need to determine a minimum value for slope using (total range-center)-deadzone jump, in this case (255-128)-77=50. So for a deadzone jump of 77, if the slope is lower than 50, our maximum inputs ( 0 & 255) won't reach maximum outputs. (In-game this means you wouldn't be able to achieve your maximum turn angle)

So, say I wanted to set my deadzone at 10 and my deadzone jump to 100:

`0x00010030 input_state_scaled
128 sub
dup sign swap
abs dup 10 gt swap
10 sub
27 mul
118 div
100 add
mul
mul
128 add `

Also, could you elaborate a bit on what in-game deadzone behaviors would be handled better by the radial approach? 

I have an idea of how I could use HID Remapper to test the game's deadzone more accurately (to determine if the radial approach would be better). I'm pretty sure it's possible, but I'm doing my best to just follow along and understand what RPN's written by others do, so I'm a long ways from getting creative. 

Since I have so many unused buttons, I was thinking I could map some of them to simulate a known numerical value increase to an axis. 

for example, Button 1 = +10 , Button 2 = +5, Button 3 = +1, Button 4 = reset, (or Button 4 = -10, Button 5 = -5, Button 6 = -1)

The joy stick would be centered, giving an output of 128. If I press and release button 1, output becomes 138. Output is persistent - the HID Remapper continues to send 138, simulating holding the stick in that positon.  If I press and release button 1 again, output goes to 148. If I press button 2, output goes to 153. If I press button 3, output goes to 154. If I press button 4, output returns to zero. 

With this, I could keep a mental count of buttons pressed and zero in on exactly where the game's threshold is, and also get a pretty good idea of the built in scaling. Can't really be sure, but I suspect I'm actually hitting my max turn angle in-game before I reach the full range of the stick. Unfortunately xbox games in general aren't going to give you that detailed information anywhere, whereas someone playing a PC game would be able to "peek under the hood" and find the relevant lines of code. Best I can do is "yeah this looks like the smallest possible circle I can fly in that's probably the max turn angle".

Sorry if it seems like I'm asking a million questions - I've been wishing I could play this game with a HOTAS/HOSAS since it came out in 2017, so I'm very excited that I am now. All thanks to you! I really appreciate all the work you put into the project, and am extremely grateful for the help you given me so far.