jfedor2/hid-remapper #258: Keyboard never lights up, doesn't show up in monitor

Schlonky, 2025-05-09:
Hi.

I'm having no luck getting this to work with a "The Shrimp Model 1" tiny keyboard. 
The keyboard works just fine when plugged in directly, but never "starts" up the lighting effect when plugged in through the remapper, and also doesn't show anything in the monitor. The remapper has been confirmed working with other devices. 

Remapper is using a RPi Pico.

I have a Wireshark capture of the device, should I just email it to you?

jfedor2, 2025-05-09:
Just post it here. Though if it doesn't even light up then it might be some hardware compatibility issue.

Schlonky, 2025-05-09:
Looks like Github doesn't like the Wireshark file, so I had to zip it. [TheShrimp.zip](https://github.com/user-attachments/files/20125557/TheShrimp.zip)

jfedor2, 2025-05-09:
I don't see anything too crazy in the descriptor.

You could try a dual Pico variant or the test build from here ("artifact" link at the bottom) that uses a newer version of TinyUSB: https://github.com/jfedor2/hid-remapper/actions/runs/14713179580

closed 2025-05-09 by Schlonky