jfedor2/hid-remapper #259: No input works after changing emulated device type

LeGittee, 2025-05-09:
Hello, 
I am running HID remapper (version r2025-03-25) on a classic Raspberry Pi Pico (single board with a soldered USB extension cord) and testing on a windows 11 laptop. After flashing the Pi Pico with HID remapper, keyboard and mouse emulation were working correctly but my intended use was with a controller so I used a browser to change the configuration (Microsoft Edge first). I set the emulated device to Stadia controller (then plugged a logitech f710 to the Pico which did not work) from this moment, the Pico did not recognize any device plugged into it. 
Of course I tried using google Chrome and setting the emulated device to "Mouse and keyboard" and "absolute Mouse & keyboard" but it did not work with mice or keyboard. 
I had another Pico so I flashed the same version of HID remapper on it and mouse and Keyboard emulation works, I loaded the configuration from this fresh Pico and saved it on the 1st one but it still didn't fix it.
I flashed the default LED blink program then HID remapper back on the Pico but I still get the same issue right away. 
I assume the issue is somewhere in the memory of the Pico ?

jfedor2, 2025-05-10:
I'm not sure what's going on, but you can wipe the flash completely by putting this UF2 on it: https://datasheets.raspberrypi.com/soft/flash_nuke.uf2

It will be as it came from the factory.

LeGittee, 2025-05-10:
> I'm not sure what's going on, but you can wipe the flash completely by putting this UF2 on it: https://datasheets.raspberrypi.com/soft/flash_nuke.uf2
> 
> It will be as it came from the factory.

I flashed this uf2 in my Pico then flashed HID remapper again but I still have the same issue.
Back when I was trying my controllers with HID remapper, I plugged an Xbox 360 controller in the Pico, could it be what caused the issue ?
Could it be the GPIO pins, is it possible they fried ?