jfedor2/hid-remapper #263: Can we use dual Waveshare RP2040 Zero to act like dual Pico version?

fakemanhk, 2025-05-13:
If I'm just looking at those GPIO connections, looks like I can also wire up 2 Waveshare RP2040 Zero according to the dual Pico version.
One reason I like the Waveshare one is, it's smaller, also it's using USB-C, wiring up 2 then I can get a purely USB-C based remapper, is this possible? Also will the dual Pico firmware work directly on it?

jfedor2, 2025-05-13:
GPIO20 and 21 are exposed on the small pads on the underside of the RP2040-Zero so they're not the most convenient to solder to, but if you can pull that off then it should work I think. We don't have a separate build right now.

jfedor2, 2025-05-15:
You can if you compile your own firmware build.