jfedor2/hid-remapper #265: Keyboard attached via rPI Pico does *not* go to sleep

gnosek, 2025-05-20:
I went through the GitHub issues and it looks like I have the opposite problem from other people :) When I connect the keyboard (Keychron K5 SE) to a USB hub, the backlight goes off when I turn off the PC (the USB hub has separate power and still provides power on its ports).

When connecting through a rPI Pico (with HID Remapper firmware), the backlight never goes off. Is there anything I can do to troubleshoot this?

jfedor2, 2025-05-20:
HID Remapper doesn't let the plugged in devices go into suspend because as you've seen it caused issues with some devices. So that's intended behavior, at least currently.

gnosek, 2025-05-20:
Understood. I'll try to experiment on my own then, -ENOTIME permitting. Would you happen to have any pointers where to start digging in the source to reenable suspend? Or has the whole feature been removed and I'd need to reimplement it?

Dzięki! :)

jfedor2, 2025-05-23:
We didn't have a suspend implementation as such, it was more of a side effect of having the downstream facing port SOF clock synchronized to the upstream facing port. So when the PC stopped sending SOF packets to HID Remapper, we stopped sending SOF packets to the connected devices and they went into suspend. But waking up didn't work properly, at least sometimes. Now we're running an independent clock for the downstream facing port SOF packets and it never stops.

braxlan, 2025-05-23:
> We didn't have a suspend implementation as such, it was more of a side effect of having the downstream facing port SOF clock synchronized to the upstream facing port. So when the PC stopped sending [...]

That's interesting! I am working on a similar project reusing your dual pico setup, and I ask myself how the SOF clocks on host and device side could be synchronized. As the code is quite complex, could you explain in short how you were/are syncing the host SOF clock to the device SOF clock so I would know where to look for? And how do you resync when the host (PC) stops/starts sending SOF packets?

jfedor2, 2025-05-23:
> That's interesting! I am working on a similar project reusing your dual pico setup, and I ask myself how the SOF clocks on host and device side could be synchronized. As the code is quite complex, [...]

They were only synchronized on the single Pico variant. As far as I can tell there's no way to influence the SOF clock on the hardware USB interface of the RP2040, though if someone knows a way please let me know.

braxlan, 2025-05-23:
> They were only synchronized on the single Pico variant. As far as I can tell there's no way to influence the SOF clock on the hardware USB interface of the RP2040, though if someone knows a way [...]

Ah okay, that makes sense, thank you for pointing out! In case I might stumble upon a solution, I will keep you informed.