jfedor2/hid-remapper #274: bug in lastest firmware r2025-06-02

jgermade, 2025-06-16:
Found that this expression does not work with the latest firmware ([r2025-06-02](https://github.com/jfedor2/hid-remapper/releases/tag/r2025-06-02)) after re plug the usb

```
0x00010030 input_state_binary 0x00010088 monitor
```

When you save to device, you can check in monitor that is working. But after unplug and replug, it does not work.

---

I am using the previous release which works propertly ([r2025-05-23](https://github.com/jfedor2/hid-remapper/releases/tag/r2025-05-23))

jfedor2, 2025-06-17:
I think I got it, the issue should be fixed in the latest release.

The issue wasn't with with monitor function but with fetching the input state in an expression.

It only happened on a Mac because the issue was triggered by a boot/report protocol mode change and macOS sets the mode to report protocol when a device is plugged in, even though that is the default (so other systems don't do it).

But I think it could also happen on a PC that uses the boot protocol mode during BIOS/bootloader stages.

jgermade, 2025-06-17:
So fast your fix, thanks!

I've found this bug by finding a way to use the remapper as a mouse jiggler that moves the cursor only after a time of inactivity. Finally got it ☺️

Btw, love this project!