jfedor2/hid-remapper #280: Keyboard input not making it to PC after r2025-01-13 and above

clrkio, 2025-06-27:
After flashing the firmware on my Pico to release r2025-01-13 (and all releases thereafter), no input is making it to my PC. However, the Monitor tab on https://www.remapper.org/config/ shows every button press. Reverting to r2024-12-05 or prior resolves the issue.

jfedor2, 2025-06-27:
Just a guess, but downgrading might work in your case not because there's been a bug between these releases, but because with certain releases, an older release cannot read the stored configuration from a newer release (because some new feature has been introduced that the old firmware version doesn't know about). So when you downgrade the firmware, your configuration is reset to the default one.

Can you try flashing the latest firmware, then **not** loading the configuration from device on the configuration website, just saving the default one to device?

clrkio, 2025-07-03:
I tried a different keyboard and it worked normally on the latest firmware (r2025-06-17). 

Is there any debug information I can provide when using the original keyboard that would be helpful?

jfedor2, 2025-07-03:
You could dump the report descriptor of that keyboard (we have [a video](https://www.youtube.com/watch?v=Qt-UlTNUnu8) on how to do it on Windows). If it shows up in the Monitor then the descriptor should be parsed correctly in theory, but we've had some quirks with that in the past.

jfedor2, 2025-10-07:
It's possibly the same issue as #295 with a Logical Maximum of -1 in the report descriptor, where it was clearly intended to be 255.

Maybe we need some generic workaround for a situation like this (ideally without breaking anything else).

```
0x05, 0x01,        // Usage Page (Generic Desktop Ctrls)
0x09, 0x06,        // Usage (Keyboard)
0xA1, 0x01,        // Collection (Application)
0x05, 0x07,        //   Usage Page (Kbrd/Keypad)
0x19, 0xE0,        //   Usage Minimum (0xE0)
0x29, 0xE7,        //   Usage Maximum (0xE7)
0x15, 0x00,        //   Logical Minimum (0)
0x25, 0x01,        //   Logical Maximum (1)
0x75, 0x01,        //   Report Size (1)
0x95, 0x08,        //   Report Count (8)
0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x95, 0x01,        //   Report Count (1)
0x75, 0x08,        //   Report Size (8)
0x81, 0x01,        //   Input (Const,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x95, 0x06,        //   Report Count (6)
0x75, 0x08,        //   Report Size (8)
0x15, 0x00,        //   Logical Minimum (0)
0x25, 0xFF,        //   Logical Maximum (-1)
0x05, 0x07,        //   Usage Page (Kbrd/Keypad)
0x19, 0x00,        //   Usage Minimum (0x00)
0x29, 0xFF,        //   Usage Maximum (0xFF)
0x81, 0x00,        //   Input (Data,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x95, 0x05,        //   Report Count (5)
0x75, 0x01,        //   Report Size (1)
0x05, 0x08,        //   Usage Page (LEDs)
0x19, 0x01,        //   Usage Minimum (Num Lock)
0x29, 0x05,        //   Usage Maximum (Kana)
0x91, 0x02,        //   Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0x95, 0x03,        //   Report Count (3)
0x75, 0x01,        //   Report Size (1)
0x91, 0x01,        //   Output (Const,Array,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0xC0,              // End Collection
```

jfedor2, 2025-10-07:
I'm still not sure I understand why this happens, but can you try this test build ("artifact" link at the bottom)?

https://github.com/jfedor2/hid-remapper/actions/runs/18323557399