jfedor2/hid-remapper #285: Chord inputs?

boxcat-chris, 2025-07-24:
Hi! Instead of just remapping 1:1 inputs, would it be possible to define chords of multiple inputs that map to a single output on key release? 

This doesn’t look possible out of the box, but it’s a feature I’d like to see because it’d allow using the remapper to convert a generic 2x4 macro pad into something like an Artseyio layout or enable a 5-key macro pad to use something like the Tap alphabet (the 5 keys are a e i o u, a+e outputs n, e+i outputs t, i+o outputs t, and so on). It looks like I could theoretically implement this with expressions, but I wanted to get your thoughts on the best way to do this before I got started.

jfedor2, 2025-07-24:
It's not as straightforward, but you can check out the "5-key chording keyboard (A-S-D-F-space)" configuration in the Examples tab.

jfedor2, 2025-07-30:
I don't think sticky works on mappings with registers as inputs.

You could reimplement it manually inside the expression. For example instead of:
```
31 recall 0.001 eq 1 store
```
try something like this:
```
31 recall 0.001 eq
1 recall not
1 recall
ifte
1 store
```

boxcat-chris, 2025-07-30:
That worked. I now have sticky shift and control keys. Thank you again!