jfedor2/hid-remapper #287: When going back to layer0 don't start sending layer0 keycodes for already pressed keys (so quickly)

mmazur, 2025-08-01:
**tldr**: if I press LayerN+Key, let go of LayerN (which goes back to layer0), but keep pressing Key, very quickly the OS starts receiving layer0's Key keycode. This behavior should be changed, as it will (almost) always cause a misfire as opposed to being an intentional outcome by the user.

**Full explanation:**

A lot of my keyboard shortcuts are optimized for being doable with just my left hand, so that I can perform them while operating the mouse. All of them are caps+key ([my configs](https://github.com/mmazur/caps-madness-keyboard)), which means my left hand usually performs them in a "rolling" motion: pinkie starts on the caps, another finger goes for a key, pinkie lifts, other finger lifts (as opposed to lifting my pinkie last, as would be "proper form").

For the past many years that I've been using [input-remapper](https://github.com/sezanzeb/input-remapper) this has never caused "misfires" (e.g. me doing a caps+e ending up with an actual 'e' getting sent to some random window) because the way input-remapper handles key combinations is that if the defined combo is e.g. ctrl+alt+t, letting go of any of the three keys will be treated as a permanent KEY_UP (until one lets go of everything else), so letting go of, say, ctrl first never leads to a plain 't' getting sent, no matter how long I'd keep holding the 't' for.

Actual example: I have caps+hjkl for arrow navigation; when I caps+h and hold that for a bit to navigate "left", it's never the case that I will let go of caps, but keep holding 'h' because I want to actually start "typing" the 'h' in the text box.

It seems to me that the "rolling" motion for such combos is natural and remapper should take that into account by either:
- never going back to layer0 on a pressed key until it's let go because it's never the case that someone wants to do layerN+Key, let go of layerN and keep holding Key expecting the layer0 keycode to be sent
- or make the 'reset' time before sending layer0 keycode so high (or at least configurably high) that a 'misfire' can never happen by accident

mmazur, 2025-08-01:
Ok, looking at the code this isn't so obvious to do, as the current layer mask is "stateless" and recalculated on each main loop iteration. So step one would need to be to add storage of previous mask (port-aware, just to make things harder) and only then try to have additional "conditional layer reset" logic.

Unless there's some clever way to do this in an easier way…

MarqueIV, 2025-08-01:
Writing this by voice on my phone so apologies for any typos.

Currently the expression support does have the ability to check the value of a key the previous time things were "processed". I use that extensively to know the difference between "the key is down" (both current and last checks are down) versus "this key was just pressed" (current is down, but last is not.)

That kind of logic may be able to solve your issue here. 

When going back to layer zero because the user releases the layer key, when scanning and determining you are on layer zero and the key is pressed, you should be able to detect was this key*just*pressed and only send the key codes then.

The only potential caveat is keys, repeating. If the calculations truly are stateless, except for the one prior instance, then the home layer may be interpreting this as a key repeat issue, not an initial click issue. A solution there would be to use the same current/last check but for the layer value itself, not just a key. That way you could use logic saying “yes, this key is currently being held down, but the layer just switched, and this key is set for ‘repeat’, so block this key until it is released (or some variation thereof.)

mmazur, 2025-08-02:
Thanks for the pointer, it does seem like it might be doable. Maybe I can code my own layer management with expressions or maybe I'll find inspiration with some of the registry examples for a different approach. 

This is unfortunately quite unintuitive for someone not versed with this kind of coding, but at least I'm Polish, so the notation's easy. ;) 

(My point that current layer resetting behavior is not ideal and should be changed stands though.)

mmazur, 2025-08-03:
Ok, so the imperfect workaround looks like this: (tried to upload a screenshot, but that fails repeatedly)

```
layer 0: caps -> layer 1
layer 1: H -> layer 2
layer 2: H -> Left arrow
```

It's not perfect, but it's better than default. If I hold caps+H, then let go of caps it keeps sending Left arrow, so there's never a chance of a misfire.
Means I have to duplicate every key definition, but that's way better than trying to encode logic for 20+ keys into an expression.

Only way of simplifying it to a single mapping per key I can think of would be if remapper had optional "filter" functions that can be specified for any mapping, where the function gets called for each key action with arguments: layer it is being called from, which key was pressed and which key is the mapping for, so that it can decide whether it should send the mapped key, not send it or do something else entirely (change a layer, which is not currently possible from an expression).
… but that'd be a major undertaking.