jfedor2/hid-remapper #292: Waveshare RP2350A USB not work on some device.

leading28, 2025-09-04:
Flash "Waveshare RP2350A USB" with the "https://github.com/jfedor2/hid-remapper/releases/download/r2025-06-17/remapper_waveshare_rp2350_usb_a.uf2",
but not work on Slimblade pro usb wired and Logitech trackman marble, but ok on ROG gladius III.
With raspberry pi pico w, all work fine.
Does anyone use this dongle? Help, Tks.

jfedor2, 2025-09-05:
@gkbd I don't think that's what's going on. I think Waveshare can't decide if they're making a USB host interface or a USB device interface, where a pull-up resistor on D+ is required. They also have them on the RP2040-PiZero board, though it's slightly more defensible there because it has a Type C port. With the RP2350-USB-A it's obviously a host port only.

This could easily be verified by wiring a USB port to some other pins on the RP2350 and checking if the devices in question now work. Though you'd need a custom build of the firmware. Or you could try removing the resistor(s) if you can find them on the board.

gkbd, 2025-09-05:
It's not host only, works in device mode too. Requires USB type A to A cable though. They have mouse demo that emulates HID device on type A port.

jfedor2, 2025-09-05:
You can wire a USB connection through an audio jack if you want (Apple did). That's neither here nor there. A Type-A receptacle is still a host port. An A-to-A cable is not permitted by the spec for obvious reasons. Not sure what you're trying to argue here.

jfedor2, 2026-04-03:
Tell Waveshare to release a version without that resistor! It's a perfect little HID Remapper board otherwise.