jfedor2/hid-remapper #295: Input keys are detected, but not reacted to or passed through

mmazur, 2025-09-06:
I have a wireless presenter pilot with a usb dongle. It sends two keys: Up and Down.
Using my latest hid-remapper loaded on a feather rp2040 (that works fine with a random keyboard I tried) I'm observing the following behavior with the pilot:
- when pressed both Up and Down are correctly detected and shown in the "Monitor" tab of the /config/ page
- the led on the board blinks when I press the keys
- however the keys are not passed through to the computer
- nor can they be remapped to something else

It seems like hid remapper is making an attempt to react to them, but somehow fails.

Any way to debug this? (Sending the physical device is an option, as I've planning on buying a new one for a while now.)

jfedor2, 2025-09-07:
Can you dump the report descriptor and ideally some traffic?

jfedor2, 2025-09-07:
I can see the input reports in the dump, but I can't see the report descriptor. How are you doing the dump?

mmazur, 2025-09-08:
Ah, report descriptor gets sent on device connect only. I redid the traffic dump, but first connected the device, then mashed some buttons.

[hs304-full.pcapng.gz](https://github.com/user-attachments/files/22207921/hs304-full.pcapng.gz)

(What I did was `tshark -i usbmon5 -w dump`, then `lsusb` and `tshark -r dump -Y 'usb.bus_id == 5 && usb.device_address == 124' -w hs304-full.pcapng`.)

mmazur, 2025-09-09:
Bought a new, "different" one – same problem. `2571:4101 HAS   Wireless Present`

mmazur, 2025-09-13:
Any luck?

jfedor2, 2025-10-07:
Can you try this test build ("artifact" link at the bottom)?

https://github.com/jfedor2/hid-remapper/actions/runs/18323557399